ImpulseRC/OSD #4: gps coordinates

fnsweet, 2018-07-28:
Are the GPS coordinates passed over telemetry using KISS FCv2 and crossfire?  I have tried to discover new senors in the taranis but no gps sensors found.

awolf78, 2018-07-28:
With latest KISS FC FW, available through latest KISS GUI here:
https://github.com/flyduino/kissfc-chrome-gui/releases

fnsweet, 2018-07-28:
I am currently running KISSFCV2F7-1.3RC36c

awolf78, 2018-07-28:
Does the GPS work on your WolfPDB? Do you see coordinates in the OSD? They are only displayed once the home position has been initialized

fnsweet, 2018-07-28:
The coordinates do work in the OSD

awolf78, 2018-07-28:
Did you upgrade the FW on the WolfPDB to the latest?

fnsweet, 2018-07-29:
The fw is the latest stable version of 17.8

awolf78, 2018-07-29:
UPDATE: Talked to KISS devs, it looks like the GPS support did not make it in the latest release - but will confirm tomorrow.

fnsweet, 2018-08-09:
Thank you, will wait for the next release :-)

awolf78, 2018-12-09:
Implemented in latest KISS FC release

closed 2018-12-09 by awolf78