ImpulseRC/OSD #5: Unify pro - Smartaudio + microphone audio

FPVlog, 2018-08-06:
Hi

I read the manual and couldn’t find anywhere if it is possible to wire the Unify so it uses both smartaudio and audio from the microphone. 

Is it possible at all? Should I jump the two pads and see if it works?

Cheers

awolf78, 2018-08-06:
Yes, jump the two pads. Solder the SmartAudio wire to the VTX TX pad on the WolfPDB. Then solder a small wire from VTX TX to the Audio pad.

In rare cases this can disrupt SmartAudio communication - then you might have to make a choice between Audio and SmartAudo. Not much we can do about it, that is a design issue of the Unify VTX. TBS recently came out with a Unify version with separate Audio and SmartAudio wire...

mortenb1337, 2019-01-03:
I had the same question, and came here. Did it work for you?

greezmen, 2019-03-28:
any plans for smart audio 2.1 support? or iam doing something wrong  its nit working with my hv pro race v2 unify thank you

awolf78, 2019-03-29:
SA V2.1 already implemented but not released yet, I can get you a beta if you like.

question2019, 2019-04-12:
hi, pls can u send me the link to the beta with SmartAudio 2.1 Support?
Thanks!

redoakleaves, 2019-04-12:
It is supported by the new firmware-version which is out since a couple of days, i guess. You should be able to update it right now.

question2019, 2019-04-12:
works perfect !!! thanks

TinoFpv, 2020-01-04:
Hi guys, noob here needs help.
I got a foxeer predator v4 on a tbs unify pro v3, the impulse ods have from left to right VTX control, audio out, so I have to bridge them?But when I do so i can't here the audio.
What to do?
Thanks and sorry for the noob question, first build.

awolf78, 2020-01-04:
I assume you are using the Apex OSD. Easiest is to strip the SmartAudio wire from the Unify a bit longer and solder it across both the VTX control and Audio pad. Please check that you have the volume turned up on your goggles. Are you sure the audio on your goggles is working?

TinoFpv, 2020-01-04:
Thanks awolf78, it was the jack not going right in.
I soldered now vtx and audio together on the apex osd.It works well.
Last question is what to do with the violet cable coming out from the foxeer.
Don’t now where to solder it :/
Many thanks for help.

awolf78, 2020-01-04:
The Violett (=Purple in Englisch) is the current sensor I guess. That one is not necessary. Make sure the VTX ground and camera ground are both soldered to the common ground pad.

TinoFpv, 2020-01-04:
Purple, exactly!All ground cables are fine.Apex osd is showing the right leds, so I think It is ok.
Thanks again awolf!

MrMPod, 2020-03-01:
> Apex OSD. Easiest is to strip the SmartAudio wire from the Unify a bit longer and solder it across both the VTX control and Audio pad.

but am I right that to change my vtx chanels I need PC computer and ImpulseRC OSD application ?
or is there way to change my vtx chanels and power without computer ?

awolf78, 2020-03-01:
No, you can change those in the field on the OSD menu. Read [here](https://github.com/ImpulseRC/OSD/wiki/Firmware:-Video-Transmitter)

MrMPod, 2020-03-01:
so is it necessary strip  the SmartAudio wire or just solder bridged the two pads VTX Control to the Audio Out pad on APEX OSD (I have already connected audio to my Unify VTX)

awolf78, 2020-03-01:
You can strip the wire longer or bridge the two pads - whatever is easier.

MrMPod, 2020-03-04:
you are right - strip the SmartAudio wire is easier than bridged the two pads :)
and it is working great, thank you very much for help