ImpulseRC/OSD #6: Wrong ESC Speed/Temp with BLHeli_32?

milindur, 2018-08-11:
Hello,

today I installed the Wolf PDB in my Reverb together with the Bardwell F4 AIO FC. The ESCs are Airbot Wraith32 v2 with latest BLHeli_32 v32.52 firmware (beta). The ESC current seems to be okay, but temperature and RPM are way off.

[image]

This sounds similar to this issue: bitdump/BLHeli#318

awolf78, 2018-08-11:
Yes, this is a known issue. I have done lots of improvements for BF support lately and I'm overdue for a new release. Need to do some more testing which is challenging with all this rain we have here...

milindur, 2018-08-11:
Thanks! Good to know about the fix, so it is not an issue of my build. If I can help testing anything regarding BF, please let me know.

awolf78, 2018-08-12:
Fixed in V19.8

closed 2018-08-12 by awolf78