ImpulseRC/OSD #7: Wrong KV and Min Voltage in Statistics

milindur, 2018-08-15:
Hello,

I am using the current firmware for the OSD (17.8) with BF 3.5.0 and Wraith32 BLHeli_32 ESC (with firmware 32.52). In the statistics after a flight the calculated KV and also the min voltage (hopefully) seems to be sometimes wrong:

[image]

In this case only stats of ESC 3 and 4 are wrong, ESC 1 and 2 seem to be fine. After another flight all ESC stats were affected.

awolf78, 2018-08-15:
Please update to 19.8 by connecting the GUI

awolf78, 2018-08-15:
Is the autonomous telemetry option disabled on all your ESCs?

awolf78, 2018-08-16:
Ok, Wraith V2 ESCs? Not the mini ones, correct?
I have the same on my 6S quad and will beat a few more packs through it. Might have to get used to this since the KV is calculated by voltage and RPM. Somehow the voltage reported by the ESC must have gone as low as 0.3V for some reason which caused such crazy high KV numbers (or negative ones because it overflowed).

milindur, 2018-08-16:
Correct, Wraith32 V2 35A. Since I am still quite new to FPV and mini quads, I primarily did cruising with low to mid throttle and only sometimes pushed throttle to max for testing. I just checked my screenshots. In all cases where the KV was too high or overflowed, the min voltage of that ESC was 0.30V. So maybe this is more an issue of the BLHeli_32 firmware?

closed 2019-12-30 by awolf78