ImpulseRC/OSD #12: No Audio just weird clicking Sounds

Richygoat, 2018-09-15:
Hello,
today i replaced the Steele PDB against the Wolf PDB an i don't have audio anymore.
vTX is the ImmersionRC Tramp.

When i disconnect the Audio wire from the PDB there is silence in the headphone. Connect it back only loads clicking sounds.

Newest firmware ist flashed.

Everything else is working fine.

regards

awolf78, 2018-09-15:
Did you solder the microphone on the WolfPDB? Can you show some pictures how you have soldered everything?

Richygoat, 2018-09-15:
Yes microphone is soldered on. And here are some pictures.

Richygoat, 2018-09-15:
[image]
[image]
[image]
[image]
[image]
[image]

awolf78, 2018-09-15:
Most probably a defective microphone. Please contact your dealer for a replacement. Sorry this happened to you.

Richygoat, 2018-09-15:
That was my thought too. But when i desolder the microphone this noise sound is still there.

Richygoat, 2018-09-15:
i could try to solder the microphone from my Steele PDB in the Wolf PDB when they're the same.

Richygoat, 2018-09-16:
> They are the same, so you could try that. No guarantees though: you could end up with two broken microphones!

I have tried the working microphone from the Steele PDB on the Wolf PDB. Still don’t working. After putting it back on the Steele PDB Everything working fine again.

awolf78, 2018-09-16:
Sorry this happened to you. Please contact your dealer for a replacement.

Richygoat, 2018-09-17:
> It could be the VTX - do you have a spare VTX you could try temporarily?

Yes. I tried 2 TrampHV. Both are working fine on the Steele PDB.

awolf78, 2018-09-17:
OK, then you need to have your board replaced unfortunately.

closed 2019-12-30 by awolf78