ImpulseRC/OSD #13: F.Port and RSSi

DerOrakel, 2018-09-30:
There is a RSSi option called Telemetry in the configurator, I hoped it would work with F.Port so I would not have to use a channel to forward rssi to the FC. But right now it does not seem to do anything, at least for my r-xsr connected via  F.Port ( KISS FC V2, latest firmware, 19.8 on my Wolf OSD, latest configurator). Also I can not find any info on that Telemetry option in the wiki.

awolf78, 2018-09-30:
The telemetry option for RSSI only works when using Betaflight. If the KISS developers make RSSI via FPort available for us then we can also support the telemetry option for KISS.

ducka, 2019-07-17:
Heya @awolf78, I’ve just received a response back from flyduino support saying that their development team is currently testing a firmware update that exposes RSSI via F.Port, and it should be available in the next Kiss firmware update.

awolf78, 2019-07-17:
Yes, and I am only waiting for Flyduino to release the new FW - we will release ours shortly after. RSSI will then automatically work for any receiver - no need to fiddle around on the radio anymore and send it back through a channel ;)

awolf78, 2019-12-30:
Implemented

closed 2019-12-30 by awolf78