ImpulseRC/OSD #14: Home arrow spinning / not working

mReXiTuS, 2018-11-08:
Hi Alex,

I've installed M8N GPS module (Beitian BN-180) and GPS is working fine, I can see coordinates in OSD and also stream them over crossfire to Taranis. However, home arrow indicator is always just spinning around and not showing actual home direction.

This GPS module doesn't have magnetometer, however according to specs, it should work even without magnetometer right? I have latest FW on both Kiss and WolfPDB.

Thanks,
Dominik

awolf78, 2018-11-08:
Magnetometer is not necessary. First lock may take a while. Nothing we can do from our side, that is up to the GPS module itself to find the satellites. The PDB just receives the GPS from the module.

mReXiTuS, 2018-11-08:
OK, I'll try to wait a bit more. Usually I had only like 4-6 satellites and I went into air. Probably that's not enough...

awolf78, 2018-11-08:
Needs at least 6 satellites for a few seconds to lock in home position.

mReXiTuS, 2018-11-08:
Ok, thx for info. Closing

closed 2018-11-08 by mReXiTuS