ImpulseRC/OSD #16: Some Display Options Not showing When Selected

awmondy, 2018-12-09:
I've searched this thread, impulserc github and googled and could not find an answer to my problem. A few options I selected to show on the display isn't on the screen. For instance, I had the vtx channel and vtx power on the screen. I removed the vtx channel and it was no longer displayed. I added the channel back and it did not show on the screen. Other display options are not showing either. I used the Impulserc app to default the pdb and had the same issue. Has anyone else had this issue?

Impulserc/Wolf fw 22.9
Kiss v1 fw 1.3-rc36c

awolf78, 2018-12-09:
Did you click Save in the ImpulseOSD GUI after you added the VTX channel back?

awmondy, 2018-12-10:
I did click save in the GUI.

FYI - RSSI, timer and GPS data doesn't show at all. I can add and remove
callsign.

awmondy, 2018-12-11:
I found what was causing the issue. I added a GPS (BN180). I connected tx to tx and rx to rx, when I should've had tx to rx and rx to tx. I reversed the connections at the pdb and the missing options were displayed on the screen. I find it odd that having the GPS connections reversed could cause some options not to be displayed.

awolf78, 2018-12-11:
It doesn't make sense for the OSD to display labels when there is no data available for them. Some pilots just plug in a GPS occasionally and then they will only see the GPS related labels when it is connected.

closed 2019-12-30 by awolf78