ImpulseRC/OSD #17: Micro Swift 3 Serial Communication Wolf PDB and Kiss v2

Ayougo, 2018-12-31:
Wolf PDB with latest firmeware
Kiss v2 with RC36I firmware
Micro Swift 3 connect via TX/RX  (TX from Swift to RX on Wolf PDB, RX from Swift to TX on Wolf PDB)

Issue: No Internal OSD menu on new swift 3 micro with serial camera control 

Yaw left, toward bottom or zero throttle, hold for several seconds, enter menu, select "camera control", Yaw Left  == No Internal OSD menu

mkct93, 2018-12-31:
Same here friend. 😐

awolf78, 2018-12-31:
Go to SETUP->CAMERA CONTROL and change CAMERA CONTROL to SERIAL.

awolf78, 2018-12-31:
Cam RX connected to PDB Camera TX?

mkct93, 2018-12-31:
Yes. I swapped it around both ways in case the camera wasn’t labeled correctly. Still nothing. I’ll try another camera tonight that I know works with the serial connection and post the results. Also I am using the rev1 pdb.

mkct93, 2018-12-31:
Fantastic! I’ll give it a go tonight. Pdb is awesome by the way. Great piece of kit. 👍🏻

closed 2018-12-31 by awolf78

mkct93, 2019-01-01:
That was it. Thank you so much.