ImpulseRC/OSD #18: OSD stopped working

Anckero, 2019-01-03:
My OSD is not showing up anymore with my WolfPDB. I don't get values nor ImpulseRC boot logo.
The green LED is on but the red one is not flashing as it should.
The PDB is not connecting to my PC.

I searched online and found on RCgroups a guy who stopped a firmware update by mistake and he had the exact same issues. So even if my problem is not coming form a bad firmware update I followed the steps.
I temporarily bridged SDA and GND then connected the USB. This trick worked, the red light was flashing and the PDB was recognized by my PC. As an update was available I flashed it and it went well.

But then I unplugged the XT60 and USB and this issue is still there on my next attempt to connect the XT60, no red LED flashing.

I'm using a KISS FC v1.03 FW1.3-RC36i

awolf78, 2019-01-04:
Do the SDA and GND trick again, but this time select Full Chip Erase in the GUI before you flash.

Anckero, 2019-01-04:
Fantastic! All is back to normal.
Thank you very much awolf78! 😃

closed 2019-01-04 by Anckero

Oberlurch3000, 2019-05-01:
Please could you tell me how this trick works? I've the same problem since today. Just no OSD in the FPV screen and the red LED is off. WIndows tells me that the device has not been recognized. Why is this happening? I did no changes to it in the last few weeks.

Anckero, 2019-05-01:
Hi @Oberlurch3000 
Here is the RCGroups thread I first found the instructions: https://www.rcgroups.com/forums/showthread.php?2806289-KISS-OSD-Firmware-new-versions-posted-here/page126#post40795311

Just follow the instructions as per Awolf78's comment on RCGroups and you should be fine. If the problem persists you may have to do the "Full Chip Erase" thing in the GUI as Awolf78 told me to do on this GitHub thread. :)

Oberlurch3000, 2019-05-01:
Wow thanks! I like things that work ;). But very strange that it happend anyway.

Anckero, 2019-05-01:
Yeah it's pretty strange. Mine stopped working like yours, without any reason. But it has been working great since this fix so you should be fine too. :)

Brightontillifly, 2021-07-10:
> Do the SDA and GND trick again, but this time select Full Chip Erase in the GUI before you flash.

this totally worked for me nice one Wolf!

Ub6ib9, 2021-08-28:
Awesome so there is hope. Out of nowhere the osd quit working and gps wasn't getting a lock. I'm hoping this process will work. I'll chime back in after trying.