ImpulseRC/OSD #20: "Battery Charged ?" show on deplay

climbhi2, 2019-01-24:
Hello,
I would like to thank you for everything you’ve done to make this OSD the best.

I'm running the latest frame ware and have been seeing "Battery Charged ?" come up blanking when I plug in a battery. When I arm the message goes off. Is this new? I have tried to isolate it by changing batteries on my goggles and transmitter. Is this just a message “battery charged?” to remind you to check the battery before you fly or have I changed a setting that I’m not aware of.
Thank You 
Tommy 

[image]

awolf78, 2019-01-26:
This is voltage based and appears when you plug in a battery that does not appear to be at charged voltage.

whig0, 2019-07-20:
This warning is shown even with 4.15 Volts. Is it possible to disable it? Or can it be adjusted somehow? 
Thanks.

awolf78, 2019-07-29:
I have made it less sensitive and dropped the warning voltage to less than 4.10V per cell. Will be included in next release.

whig0, 2020-09-12:
Unfortunately I still have the problem. The warning shows for me as soon as the voltage drops under 4.1V Per Cell. 
Is there still no possibility to just disable it?

awolf78, 2020-09-12:
The warning disappears as soon as your arm. It is just a reminder that your battery may not be fully charged - which is the case at 4.1V.
Alternatively you can add a voltage offset and raise the voltage by e.g. 0.1V in the [Battery menu](https://github.com/ImpulseRC/OSD/wiki/Firmware:-Battery#battery-menu)

whig0, 2020-09-12:
Yes I did that already. But the warning appears as well just with .1V offset at 4.0V :-)
It even shows when not armed.

awolf78, 2020-09-12:
But it disappears when you arm? It is just a reminder that your battery is not fully charged...

whig0, 2020-09-15:
Sorry for late reply. 
No it does not disappear. And it's in the middle of the screen. 
[image]

awolf78, 2020-09-15:
Hm, that is very odd. Can you please describe your setup and how you have it wired up? Do you see a message ARMED or DISARMED when your arm/disarm? What version ImpulseOSD is on there?

whig0, 2020-09-16:
It's wired like in the Philipp Seidel Wiringplan. 

Impulse OSD Version is V86.25
Kiss FC v2 (1.3RC45y)
4x Kiss ESC 32A
TBS UNIFY Pro32 VTX @7.5V
Foxeer Falkor Cam @5V
No Mic, No GPS, No LEDs
Spektrum Receiver
4S LiPos

Yes, I get the Armed and Disarmed messages.

awolf78, 2020-09-16:
Can you try flashing again?
If that doesn't help I would need a backup of your ImpulseOSD and KISS FC configuration...