ImpulseRC/OSD #22: New pinouts vor VTX and cam on rev2

benazet, 2019-03-08:
Hello I just found out that the pinouts have changed for VTX and cam, and it is not mentionned in the hardware section of the github.
As it seems to be the official user guide for the Wolf, i think it would deserve a warning or an update.
I found the info on a product info page of a merchant site : https://www.drone-fpv-racer.com/pdb-osd-wolf-reverb-impulserc-3407.html 
I think this picture has its place on the hardware page.
Although it was annoying, luckily nothing was broken but i guess this is a tribute to the quality of the product. 
Now i switched the wires it works great.

awolf78, 2019-03-08:
I literally just updated the hardware section a few minutes ago showing the new pinout. Sorry for the inconvenience.

benazet, 2019-03-08:
Great job !

I guess i have to blame the merchant site for delvering to quickly then ! I ordered it monday and began to build wednesday.

benazet, 2019-03-08:
The link of your picture is broken.

awolf78, 2019-03-08:
Fixed

closed 2019-03-08 by awolf78

benazet, 2019-03-08:
;)