ImpulseRC/OSD #23: LED Strip + WolfPDB + Kiss FC v2

karat74, 2019-03-27:
Could you help me find the way to have possibility of turning off/on LED strips connected to Wolf PDB using my Taranis QX7S switches or knobs?

awolf78, 2019-03-28:
Setup of the LED strip is only possible through the OSD menu currently, you can find more information [here](https://github.com/ImpulseRC/OSD/wiki/Firmware:-LED-strip).
You currently can turn them on/off with the arm switch.

karat74, 2019-03-28:
awolf78, thanks.

rauar, 2019-12-24:
+1 for having more options than just the arm switch. Dedicated Aux channel assignment would be great. Even better: multiple levels (e.g. 5 levels between 1000-2000ms)

1000-1199
1200-1399
1400-1599
1600-1799
1800.-2000

awolf78, 2019-12-30:
I can do something similar like I did with THROTTLE: You define two colors and depending on the AUX position (could be a switch or knob) it will calculate a gradient between those two colors (via euclidean distance).
To turn off the LEDs during flight you can simply set (0,0,0) for COLOR1. 

Advantage of this approach would be that you can assign AUX channels for each channel individually. Disadvantage is that you cannot turn off all the LEDs regardless of their assigned function. That would require an additional AUX channel setting for the LEDs. However this will cause confusion since you would have AUX channels available for the LED functions AND a global AUX which would shut the LEDs off. This is getting complicated, not a big fan of that. Trying to keep it KISS.