ImpulseRC/OSD #25: No Connection to FC

dawiddi, 2019-03-31:
Hello there,
I just build my first wolf pdb setup with a brand new board and a Kiss FC V1. After updating all firmwares (Kiss on 1.3RC36 Stable and the latest OSD firmware) and do all settings (Dummy VTX, No Logger/OSD) I have the following issue.
After booting up every time the message "Detecting FC Type" shows up. Sometimes after a while, the message just disappears, other times it follows "No connection to FC" and the OSD hides. I already checked the port cable, it seems to be everything ok. I resoldered the cables to the VTX and the cam aswell. The Kiss and the PDB are brand new.

I've read all the issues related to my problem, but could not find a solution. Can you help me?

Thanks a lot.

Hydromaster1, 2019-04-03:
I am having the same issue. Kiss FC V2, tramp vtx, & crossfire. The problem goes away when GPS is disconnected. Using an RDQ  M8N GPS. GPS data appears and disappears.
Thanks for all the hard work and awesome firmware!!!

awolf78, 2019-04-04:
I just released a fix for this issue, please connect GUI and update

Hydromaster1, 2019-04-05:
Update complete and the issue is gone. Only thing I noticed was the home arrow spins continuously.
Could this be due to close proximity to the home point. I only did one test flight about 300 meters out.
Hope to get another flight in tomorrow. Many Thanks!!

awolf78, 2019-04-05:
If the home arrow keeps spinning then the GPS has not been able to establish a fix yet. You can turn the satellites label on which will show you how many satellites the GPS has found. For a first fix it can take several minutes...

Hydromaster1, 2019-04-05:
Got it.
Many thanks!!

dawiddi, 2019-04-05:
Many thanks. I couldn‘t try flying yet, but the message disappeared in standby.

dawiddi, 2019-04-07:
So I got the time to test it today. But unfortunately the voltage, power and mah values were way too high (330 V, 150 A etc.) and after a short while it lost the connection to the FC. What should I do?

awolf78, 2019-04-07:
Can you revert to V24.11 and see if you still have the same issue?

dawiddi, 2019-04-13:
Hello,
I just got the chance to test it today. The wrong voltage values are gone, but instead the "Detecting FC Type" and "No connection to FC" comes back after a few seconds of flight.
Any further suggestions?

awolf78, 2019-04-13:
If you reverted to V24.11 and you get a "Detecting FC Type" then there is definitely something wrong with your board or wiring. Do any of the chips or regulators get very hot shortly after you plug in? Can you post some pictures of your wiring?

dawiddi, 2019-04-14:
[image]
[image]

I already measured the wiring to the Kiss FC, all cables are okay. When I touch the PDB after powering, it stays cool. It seems good. But my Kiss FC gets very hot near the USB port. I thought, that would be normal because it is a new one.

I attached some pictures of my wiring.
[image]
[image]

awolf78, 2019-04-14:
Wiring seems OK. What type battery are you running? 4S, 5S, 6S?
VTX is connected to HV power - not sure which VTX you are using.
Did you change anything in your serial configuration of your KISS FC. Check in the KISS GUI Advanced tab. Please post a screenshot of that.

dawiddi, 2019-04-14:
Battery is 4S, I use a Unify Pro HV, that works great.
I attached a screenshot of the advanced tab.
[image]

awolf78, 2019-04-14:
Strange that your Yaw Filter is at 92 - should be at 35 or 40. But that should not matter for the FC communication. Do you have another FC you can try?

antantantantant, 2019-05-03:
Hi!
I have this same 'detecting fc type' - 'no connection to fc' problem with wolf pdb + matek f405-std combo. PDB seems to work fine, it has updated itself to v40 something, fc is working and is configurable in betaflight app, it's connected to a crsf receiver and reacts to controls, I can arm (and probably fly) the quad, but can't control the PDB with stick commands or get BF OSD items to show on screen etc.
I've attached some images of my wiring (sorry for the mess :) ). I'll make more photos if you need to see details, just point me where you need to look.
Any suggestions appreciated :)
Thanks!

[image]
[image]

awolf78, 2019-05-04:
Did you read [https://github.com/ImpulseRC/OSD/wiki/Betaflight-WolfPDB:-Please-read!!!](url)?
Configured MSP for the UART you are using for FC communication to the PDB?

antantantantant, 2019-05-04:
Believe it or not, I missed the article, although I searched for instructions :/ Sorry!
Anyway, just set up BF as instructed and everything works now :)
Thanks!

AiroFpv, 2019-05-11:
So i have also been having this same issue, new board, connect to gui and flash newest fw. When i first plug in, osd is fine and i can access menus for a bit then "no connection to fc" shows up. Same if i go for a fly, after a few seconds, osd goes away and no connection to fc. Any tips?

AiroFpv, 2019-05-11:
Link on your last reply doesnt work either. Cant read the please read!

awolf78, 2019-05-11:
[Please read](https://github.com/ImpulseRC/OSD/wiki/Betaflight-WolfPDB:-Please-read!!!)

awolf78, 2019-05-11:
Can you describe the setup you are running? FC type, lipo voltage, etc.

AiroFpv, 2019-05-11:
Sorry, i deleted the link comment cause i am running kiss v1 fc and 32a escs, 5v unify, 4s with caps on esc's in preparation for 6s transition.

awolf78, 2019-05-11:
What FW are you running on the KISS FC?
Not sure if you have tried to run 6S on your rig, but the KISS V1 cannot handle 6S lipo voltage. Most of the time the voltage regulator gets damaged if you try to run 6S and although it still flies you get these connection issue. Not 100% sure that is the issue, just one possibility.

StevenFpv, 2019-09-24:
ciao ragazzi, ho lo stesso problema appena collego la lipo mi rimane il logo impulse con la scritta DETECTING FC TYPE.  una build che funzionava, ho sostituito con una nuova kissv2 e nuova ricevente r9 slim plus, ho aggiornato con l 'ultimo fw di impulse e l ultimo su gui, aggiungendo il GPS. ho staccato il gps e provato fw precedenti ma non sono riuscito a risolvere il problema.

vladsubbotin44, 2019-10-29:
pdb wolf writes in OSD Detecting FC Type. flight controller mamba f405 mk2. whats problem ?

sblakemore, 2019-10-30:
@vladsubbotin44 please ensure that you have updated to the latest version of ImpulseOSD. Can you also check that you have MSP enabled for the correct serial port in BF and that you have soldered RX to TX and TX to RX.

AiroFpv, 2019-11-09:
So, something really weird just happened, went to update the osd, run
impulseosd plug in, says do you want to update, click yes then error
message saying lost connection try power cycling and re run gui. Now when i
plug in the computer it says that device isnt working correctly, impulseosd
gui just sits there and doesn't see it and impulserc driver fixer doesnt
see it either. No flashing heartbeat, should i try to solder sda and gnd or
did it fry?

awolf78, 2019-11-09:
[https://github.com/ImpulseRC/OSD/wiki/Rescue-flash](url)

blackdrone911, 2020-01-12:
> If you reverted to V24.11 and you get a "Detecting FC Type" then there is definitely something wrong with your board or wiring. Do any of the chips or regulators get very hot shortly after you plug in? Can you post some pictures of your wiring?

I have the issue with my PDB v2. The small voltage regulator is getting hot shortly after plug-in the LiPo. Also it is not possible to connect the board via USB. Is the board defect?

awolf78, 2020-01-13:
Last rescue attempt would be to attempt a rescue flash. You would have to temporarily bridge the SDA and GND pads. If the red LED doesn't blink though and you cannot connect with USB it doesn't look good :(
Instructions for rescue flash are here: https://github.com/ImpulseRC/OSD/wiki/Rescue-flash

StevenFpv, 2020-01-13:
Solved

Ottieni Outlook per Android<https://aka.ms/ghei36>

blackdrone911, 2020-01-13:
In my case, unfortunately not.
Thanks for the instruction awolf78. But now I am sure that I can only keep the board alive, if I change the voltage regulator. Today I ordered some of them. I hope the replacement will be the solution. 
Today I tried an experiment. I put some ice in a plastic bag on the “hotspot” and voila, there was connection via USB until the regulator got too hot again.
But, the question is, how it can get damaged? The board is nearly brand new.
Only flown 6-8 lipos with it.

awolf78, 2020-01-14:
Unfortunately these failures can happen even though we tested the board. We do not make the voltage regulators ourselves. As an alternative you can contact your dealer and request a replacement if you do not want to replace the regulator yourself.

Frakovich, 2020-01-20:
Hello same problem here. "No connection to FC" and the OSD hides. I already checked the port cable, it seems to be everything ok. I resoldered the cables to the VTX and the cam aswell. The Kiss and the PDB are brand new. I tried the trock : bridge the SDA and GND pads. But the red LED doesn't blink though and I cant connect with USB... Even after desolder everything...

awolf78, 2020-01-20:
Can you post a picture how you have it wired up?

Frakovich, 2020-01-20:
Thx for your fast answer :) 
Shame on me...  Im actually at work lol (my setup is at home). I take a picture after my job... But even after unwire all ... I have still the problem

Frakovich, 2020-01-20:
> With everything but Lipo power disconnected: Does the red light blink at all? With or without SDA-GND bridge?

Nope only green led :( ... Here is a pic
[image]

awolf78, 2020-01-20:
Do you have a multi-meter? Can you check for continuity between Lipo +-?

Frakovich, 2020-01-20:
Multimeter show 15.2 (4s for test)

[image]
Yes i take

Frakovich, 2020-01-20:
Kiss fcV2 is ok  (brand new)
esc ok (brand new too)
Crossfire ok (brand new)

Frakovich, 2020-01-20:
[image]

Red wire , dont touch black wire i carefully check this on magnifying glass :)

Frakovich, 2020-01-20:
> 
> 
> Do you have a multi-meter? Can you check for continuity between Lipo +-?

[image]

Red wire , dont touch black wire i carefully check this on magnifying glass :)

If it that you want to know

awolf78, 2020-01-20:
I was trying to determine if there is a short on the PDB somewhere - if there is not it will help your warranty claim even further. The KISS FC receives direct Lipo power, so that is not affected. 
Anyhow, please contact your dealer for a replacment. I'm sorry this happened to you.

awolf78, 2020-01-20:
> Multimeter show 15.2 (4s for test)
I did not mean voltage, but continuity.

Frakovich, 2020-01-20:
> 
> 
> > Do you have a multi-meter? Can you check for continuity between Lipo +-?
> 
> [image]
> Multimeter show 15.2 (4s for test)

> 
> 
> I was trying to determine if there is a short on the PDB somewhere - if there is not it will help your warranty claim even further. The KISS FC receives direct Lipo power, so that is not affected.
> Anyhow, please contact your dealer for a replacment. I'm sorry this happened to you.

Thanks for your precious help ! My dealer is cool i think he will change the board

Frakovich, 2020-01-20:
> 
> 
> > Multimeter show 15.2 (4s for test)
> > I did not mean voltage, but continuity.

Sorry i didnt understand ... when i do the test there is no bip ...

Frakovich, 2020-01-20:
> 
> 
> I was trying to determine if there is a short on the PDB somewhere - if there is not it will help your warranty claim even further. The KISS FC receives direct Lipo power, so that is not affected.
> Anyhow, please contact your dealer for a replacment. I'm sorry this happened to you.

Thx for your help !

Frakovich, 2020-01-22:
Hello, my dealer is ok to replace my pdb. they will check our discussion. I already order a new one ... As i say : shit appens... No luck I hope the next pdb will be ok !

blackdrone911, 2020-01-22:
Glad to hear it. In my case, unfortunately, I should have to send in the PDB to the USA. This is of course associated with immense circumstances here in Germany.

DrewR1, 2020-01-31:
I've had the same problem and spent 3 evenings this week connecting it in every way under the sun to a matek f722 because I use Crossfire so the SPI will not work for my setup. Just using the OSD alone and not the WolfPDB. Anyhow, I give up on this product as the red blink has stopped at this point even though it still boots up and searches for an FC only to say "no settings received from FC 1." Has anyone in this thread found resolution?

awolf78, 2020-01-31:
Did you read this: https://github.com/ImpulseRC/OSD/wiki/Betaflight-WolfPDB:-Please-read!!!

DrewR1, 2020-02-01:
Yes, and I just did it again to be sure and still get "No connection to FC."

awolf78, 2020-02-01:
What version BF are you using?

DrewR1, 2020-02-01:
Using 3.5.6 (what the Matek F7 comes flashed with). 
[image]

But that image is incorrect as the labeling of "FC RX" and "FC TX" are reversed. Now the OSD showed up for about 8 minutes and then just vanished? Rebooting or different battery do not bring it back. Is this a special feature? How do you get it to stay on?

awolf78, 2020-02-01:
Does the red LED still blink?

DrewR1, 2020-02-01:
NO, not blinking

awolf78, 2020-02-01:
:(
[Rescue flash](https://github.com/ImpulseRC/OSD/wiki/Rescue-flash)

DrewR1, 2020-02-01:
I will say, DANG it's good when it's on :) :) :)

DrewR1, 2020-02-01:
Will do and thank you graciously for the assist.

DrewR1, 2020-02-01:
Okay, it is working, not overheating and no red led. My guess is that the red led is broken or burned out....

[43 more replies]