ImpulseRC/OSD #26: Foxeer Predator V3 mini OSD + Futaba RSSI

DrHasch, 2019-05-26:
Hello there,

I connected my **Foxeer Predator V3** to the PDB but i am not getting into the menues. Hooked the  OSD pad to the OSD pin of the cam and the ground pin to the pad of the supply. I tried alot of settings but none of them worked.

As i wanted to use my old **Futaba T7C** i got myself a **ElgaeRX E6208 8CH Mini** receiver from Bangood, which is working fine and does his job. As it has a **RSSI** pad i wanted to ask whether I can connect it directly to the RSSI in pad of the PDB?

Thanks alot,
Thomas

awolf78, 2019-05-26:
https://github.com/ImpulseRC/OSD/wiki/Firmware:-Camera-control#troubleshooting

https://github.com/ImpulseRC/OSD/wiki/Firmware:-RSSI

DrHasch, 2019-05-26:
Already looked up this pages but it did not work

awolf78, 2019-05-26:
Did you wire up the OSD wire of the Predator camera? How did you wire it up? Can you post a picture?

RSSI: Yes, try connecting it to the RSSI pad on the WolfPDB. Then you have to see what readings you get and adjust min and max accordingly.

DrHasch, 2019-05-26:
[image]
Gnd of the camera (black cable) to the gnd pad on the supply of the camera on the PDB. White cable to the OSD pin on the board.

For the RSSI, where can i look up the readings?

awolf78, 2019-05-26:
Try removing the gnd wire coming from the OSD plug on the camera. That is not necessary anyhow since you already have the camera gnd connected.

RSSI: I don't know what RSSI values your receiver will output - that could be different even if you have two of the same. You will have to look at the values in the SETUP->RSSI screen in the OSD while keeping your radio close (about 1m or 3ft distance) or move it far away (or turn it off).

DrHasch, 2019-05-27:
So i managed to get into the camera control. Somehow it dit not work when i had the setting on predator.

The RSSI pin provides a voltage of 3V when connected and 0V when the remote is switched off. But the RSSI IN pin on the PDB does not really read it.

awolf78, 2019-05-27:
Did you set the RSSI channel to "ANALOG" in SETUP->RSSI? On the bottom of the screen you can see the values.

DrHasch, 2019-05-27:
Yes i did. The value just jumps from 0 to maximum 3 not more... But on 3V it should be like 2000? Am i right?

awolf78, 2019-05-27:
Connect GUI and update to V44.12.
Looks like I picked the wrong ADC channel for the RSSI input - sorry about that.

DrHasch, 2019-05-27:
No problem ;) thanks for the help

closed 2019-05-27 by DrHasch