ImpulseRC/OSD #27: GPS not detected

Anckero, 2019-06-05:
Hi,

I'm having a hard time making my BN-180 module work with my Wolf PDB.
Here is my setup:
- KISS FC v1.03 FW1.3-RC36i (latest update available)
- Wolf PDB  Rev1 FW45.12 (latest update available)
- GPS BN-180

The GPS is wired correctly to the Wolf PDB, it flashes in blue and after a bit of time it flashes in red too (meaning it got a GPS fix).
Just in case, here is my wiring:
BN-180 red wire (3.0V-5.5V) → "5V" pad of the PDB (on top of the "GPS TX" pad)
BN-180 green wire (RX input) → "GPS TX" pad of the PDB
BN-180 white wire (TX output) → "GPS RX" pad of the PDB
BN-180 black wire (GND) → "Gnd" pad of the PDB (On top of the "Mag SCL" pad)

I turned on the item "GPS coordinates" in the OSD but there is no data shown when I leave the OSD settings.

Could you tell me what I missed?
A friend with a Kiss FC v2 and Wold PDB is facing the exact same issue with a BN-220 which is exactly the same hardware as my BN-180.
I saw, here on GitHub, that you (the ImpulseRC devs) used a BN-180 in the development process so I really don't get why mine is not recognized.

Thanks in advance

awolf78, 2019-06-07:
Can you post some pictures of your setup please? I also need to know what else is connected to the PDB and how.

Anckero, 2019-06-10:
Hi awolf78,

I brought my setup to my work place and a colleague from the electronic team checked everything. We tried to wire up everything again and it worked.
The only difference was that I twisted the wires in the first place and my colleague told me that the power wires could create problem on the RX and TX wires. Now with strait cables the Wolf PDB and the GPS are talking to each other.
I will close this issue then, sorry for the inconvenience. :)
I didn't know that twisted wires could have this kind of effect.

Thanks for your support!

closed 2019-06-10 by Anckero