ImpulseRC/OSD #30: Cant connect to my wolf pdb with ImpulseOSD app in windows

charalambos12, 2019-09-19:
After updating to the last version of impulseosd app in windows i cant connect to my wolf anymore and i have the message in the attachment. What can i do?
[image]

awolf78, 2019-09-20:
Disconnect everything, reboot your PC, try again.

charalambos12, 2019-09-20:
Did it many times. Problem did not solved. Before updating the app everything was working perfectly.

awolf78, 2019-09-20:
Bridge SDA and GND pads temporarily. Plug in Lipo. Remove SDA and GND bridge. Connect USB and GUI and flash again, but this time click on "Full chip erase" before you flash.

charalambos12, 2019-09-20:
Thank you for your support!

charalambos12, 2020-01-27:
hello, after doing the procedure, at the end when i open impulseOSD program, i have a new message saying unexpected error, power cycle the device and I cannot continue with the re flashing. I have the same problem with two wolf pdbs v1 (they are working perfectly, except the connection with gui). Thank you in advance.

awolf78, 2020-01-27:
Do you have a different Windows PC or Laptop you can try?

charalambos12, 2020-01-27:
I tryied with mac laptop and I successfully updated the pdb to the latest version.