ImpulseRC/OSD #31: Wolf PDB won't connect to Impulse OSD app

briankalski, 2019-09-23:
I can't get the Impulse OSD app to see my Wolf PDB.  I did flash the Steele PDB firmware to the board which I am assuming is the wrong firmware.  Is there somewhere I can get the Wolf PDB firmware?

awolf78, 2019-09-23:
How did you flash the SteelePDB firmware on the WolfPDB?

briankalski, 2019-09-23:
Using Xloader.  It was my first quad.

awolf78, 2019-09-23:
XLoader cannot flash the WolfPDB. Does the WolfPDB work (red LED is blinking and shows OSD)?

briankalski, 2019-09-23:
Everything works fine.  The OSD works.  I just can't connect to ImpulseOSD to change my OSD settings.  I ran the driver fixer and that won't connect either.  I tried a Mobula7 and it worked with that.

briankalski, 2019-09-23:
The last time I changed the OSD settings I had to unplug port 2 and flash the Steele PDB config tool.

awolf78, 2019-09-23:
The ImpulseOSD GUI does not work with the SteelePDB. You will have to flash the config tool if you want to make changes to your OSD.

closed 2019-09-23 by awolf78