ImpulseRC/OSD #32: WOLF REVERB PDB OSD Not Recognizing KISS FC v2, multiple quads on powerup.

Struggen, 2019-09-24:
I have 2 Reverb's, both running KISS FC v2 & Wolf PDB rev 2 on latest firmwares that are experiencing failures to recognize the FC when the battery is connected. Reconnecting the battery temporarily resolves the issue. It is occurring every almost every other battery. The problem started occurring after I updated the PDB and KISS GUI's and then updated the firmware for both.

ReverbAlpha
Reverb 5"
Wolf PDB/OSD Rev 2
KISS FC v2
KISS 25amp ESC's
Frsky X4R-SB
ImmersionRC Tramp HV
RunCam Eagle 2
BN-220 GPS

ReverbBeta
Reverb 5"
Wolf PDB/OSD Rev 2
KISS FC v2
KISS 32amp ESC's
25v caps on ESC's
Frsky X4R-SB
ImmersionRC Tramp HV
Foxeer Predator Mini 3
BN-220 GPS
LED's on the FC powered by 5v Diatone regulator

awolf78, 2019-09-25:
Can you please update to V55.15 I just pushed?

Struggen, 2019-09-25:
I have updated one quad to 55.15 and have not had the issue occur during bench testing.

closed 2019-10-24 by Struggen