ImpulseRC/OSD #33: OSD will not centered

carloscampos775, 2019-10-12:
Can't get APEX OSD to center vertically using 57.15. Can't move items to the top of the screen. Using RunCam Swift Mini 2 with TBS Unify Pro 5v

sblakemore, 2019-10-13:
What display device are you using, goggles or screen of some kind? Different displays often show different portions of the image.

Can you please get some DVR footage so we can see how things look?

Might also be worth doing a flash with full chip erase enabled to make sure the config is clean.

carloscampos775, 2019-10-13:
I'm using Fatshark HDO’s with RapidFire. I'm not having this problem with the Wolf Reverb PDB only the Apex OSD. It's also cutting off the back/exit item in the OSD menus. I will try the full chip erase and update the issue.

[image]

carloscampos775, 2019-10-13:
This is after a full chip erase. Still not working correctly. I have also tried another camera but the problem still persists.

Default OSD layout after full chip erase:
[image]
 
This is what I need to do in the configurator to get the OSD items to show on screen:
[image]

Herbiefpv, 2019-10-14:
I am also having this exact issue. I can adjust the osd side to side but it will not let me move it up. I can see something barley at the bottom. I also tried re flashing but no luck.

Herbiefpv, 2019-10-14:
> Thanks for your help guys.
>
> I was tied up working on something today but will be able to look at this
> properly tomorrow. I'll let you know how we get on.

Awesome thanks Sean!

sblakemore, 2019-10-15:
Thanks again for the feedback, I believe I just fixed this. We should have a new release soon for you to try!

sblakemore, 2019-10-15:
If you would like to test a Beta release with the fix I can enable your OSD for Beta updates. All you need to do is connect the OSD to the GUI and then press Ctrl+Shift+D and you will get a popup with your hardware ID. Then just hit Ctrl+C to copy the ID and paste it in here and I'll set it up for you!

carloscampos775, 2019-10-15:
I would like to try the beta. Here is my hardware ID

Device ID: 0011 1025 4847 750A 2037 7341

awolf78, 2019-10-15:
Please download the beta GUI from here:
[http://osd-storage.impulserc.com/deploy-beta/setup.exe](url)

That should offer you V59.15. Let us know how it goes.

carloscampos775, 2019-10-15:
Looks much better. If you could get it to center just a little high that would be perfect.

Default look after flashing
### 
[image]

Herbiefpv, 2019-10-15:
> Please download the beta GUI from here:
> http://osd-storage.impulserc.com/deploy-beta/setup.exe <http://url>
>
> That should offer you V59.15. Let us know how it goes.

O was i suppose to use a certain gui? I just used the latest one i got the
other day but it will centered mine lol

carloscampos775, 2019-10-15:
I was able to center the OSD by using the Advanced options. However, I have gotten a flickering OSD twice now since updating  

Video of issue:
https://www.dropbox.com/s/xo40cu8j2zytp4t/PICT0051.AVI?dl=0

sblakemore, 2019-10-16:
@Herbiefpv Beta GUI not a requirement for this update, you will be fine :)

@carloscampos775 Vertical jumping like that is to do with the sync level your camera is putting out. You should be able to fix it by adjusting the sync detection level using the camera settings dialog in the GUI. There is a slider in there you can move up or down, will take some trial and error to find a value that works for you. Looks like it's about on the edge so probably will only take a small movement up or down to be stable.

carloscampos775, 2019-10-16:
@sblakemore I found the camera sync setting. I think i'm good. Thanks for your help ;)

closed 2019-12-30 by awolf78