ImpulseRC/OSD #34: Apex OSD camera control w/ Foxeer Predator v4

bosbefok, 2019-10-23:
This camera has Day, Night, LED, BW mode presets which you can change by long pressing the right button on the remote control board.  I'm having an issue where camera control automatically changes to the next preset when you enter the camera control menu.

All camera control settings are stock with reference voltage of 2.33v. 

Any suggestions?

awolf78, 2019-10-23:
What happens if you are not in the camera control menu? It does not change the Day, Night, LED, BW preset then?
The default reference voltage is 3.3V, not 2.33V

bosbefok, 2019-10-23:
It only changes mode when I enter the camera control menu. Sorry, ref voltage is set to 3.30v.

bosbefok, 2019-10-27:
Is this something you can fix in the next update?

awolf78, 2019-10-27:
Not 100% sure yet, but we think we came up with a possible fix for this.

bosbefok, 2019-10-27:
Great. Let me know if I can assist in any way with testing.

SRWieZ, 2019-10-31:
I have a similar problem. My camera is constantly changing mode all the time whatever I do. 
When I enter in a menu it constantly trying to go right. (in both menus, external and internal osd of the camera)

Board : Apex OSD
Camera : Runcam Micro Swift 3
Ref voltage : 3.2

bosbefok, 2019-11-08:
Hi Alexander, any progress with this issue?
I've noticed sometimes the mode is being changed now just when I open the OSD menu, even without accessing camera control menu. Doesn't occur every time but has happened on a number of occasions.

awolf78, 2019-11-08:
Unfortunately this is not going to be an easy fix since the chip does not behave like we were expecting. We have not given up on this yet, but a solution for this may take quite a bit longer. 
As a workaround you can get a Runcam Camera Control board and set the camera control in the OSD to "serial". That will work for sure and not have any mode changes.

bosbefok, 2019-11-08:
Will setting the camera control to serial also disable the camera control functionality if it is wired for analogue? I've resorted to just pulling out the OSD wire from the camera for now, but if this is a way to do it in the OSD settings it would be better as I could enable when needed, and the disable once changes are made.

awolf78, 2019-11-08:
No, unfortunately not. The OSD wire from the camera must be connected to the Runcam camera control board for this issue to go away. Otherwise it is better to keep it disconnected.

AiroFpv, 2019-11-21:
Im having the same problem where it switches all the time. Running the johnny edition runcam. Still no fix?

awolf78, 2019-11-21:
Your best chance is using a Runcam camera control board and set the camera control output on the OSD to serial.

awolf78, 2019-11-28:
OK, possible workaround which works on the JohnnyFPV Runcam Swift Mini 2: Set the "Camera Sync" level to 200. 
You can set the "Camera Sync" level in the GUI (click on the three stripes on the top left and then on Camera setup) or in the OSD menu (yaw left and hold for more than 3 seconds, SETUP->CAMERA CONFIG->CAMERA SYNC.

dweight, 2019-12-10:
I'm having the same issue with runcam swift micro v3 - your workaround helps but I can't change any settings now. This camera supports uart control is it possible to switch it on with the osd board? How should I wire it then?

awolf78, 2019-12-10:
Leave the wire where it is, but it to serial in the menu: SETUP->CAMERA CONFIG->CAMERA CONTROL and set it to SERIAL. You may have to change to serial mode on your camera as well, please check their documentation.

awolf78, 2019-12-30:
The preset switching issue has been fixed with V65.18 (not released yet when I wrote this). I am locking this conversation for others that may have the same issue. If you still experience issues with camera control, please open a new issue.