ImpulseRC/OSD #38: Led Strip Not Changing Color

dsalexander99, 2019-11-12:
I installed the led strips provided with the apex frame and they only blink red. I cannot change the color through the apex osd.

awolf78, 2019-11-13:
Please read [https://github.com/ImpulseRC/OSD/wiki/Firmware:-LED-strip](url)

dsalexander99, 2019-11-13:
I’ve already read the info provided. I go thru the osd menu to led strip and try to change mode and color but nothing happens. It’s just stuck blinking red in the “knight” mode.

awolf78, 2019-11-13:
Can you post a DVR what you are doing?

dsalexander99, 2019-11-23:
https://www.dropbox.com/s/7y3exxyjwp1n5ge/Video%20Nov%2014%2C%208%2012%2021%20AM.mp4?dl=0

dsalexander99, 2019-11-23:
That’s a link to the video. Bare with me, I couldn’t get my dvr to record.

awolf78VDRS, 2019-11-23:
LED #1 is the one on the regulator board. LED #2-#7 are the ones that are in Knight mode. You need to select those and change the function/color for them

awolf78, 2019-11-23:
LED #1 is the one on the regulator board. LED #2-#7 are the ones that are in Knight mode. You need to select those and change the function/color for them

dsalexander99, 2019-11-23:
Thanks, I figured I was doing something wrong just didn’t know what.

closed 2019-11-23 by dsalexander99