ImpulseRC/OSD #40: No audio only SmartAudio

dsalexander99, 2019-11-24:
I can’t get any audio from the onboard mic. I have the unify pro 5v with the audio cable jumped across the audio and vtx control pad on the osd. I can control my vtx but I have no sound. Any ideas?

awolf78, 2019-11-24:
Do you have audio on any other quads you have that works?
Are you sure the volume on your goggles is turned up? Does the earpiece work you use?

dsalexander99, 2019-11-24:
Goggles are the FatShark attitude v5 which have the volume on high from the factory. The audio worked when I first built it then I ran into the problem that I couldn’t change vtx settings through my osd. So I jumped the audio and vtx pad and that gave me the ability to control my vtx but lost audio from the mic onboard. And yes I’ve tried different earbuds and they all work on other builds.

dsalexander99, 2019-11-27:
Am I just gonna have to choose between vtx control and audio ?

sblakemore, 2019-11-27:
@dsalexander99 this Issue tracker is really only meant for software bugs with ImpulseOSD. The problem you are having is not something which can be fixed in software.

You can try contacting ImpulseRC or TBS support directly as what you are trying should work fine and does for many people. Either there is something wrong with your hardware or there is something which can be fixed with troubleshooting via support.

TBS have a support page about this, it might be that you need to add a 100ohm resistor when bridging the pads. The 100nF capacitor is already on the OSD PCB. http://team-blacksheep.freshdesk.com/support/solutions/articles/4000100084-tbs-smartaudio-what-it-is-and-how-it-works

closed 2019-11-27 by sblakemore