ImpulseRC/OSD #43: Wolf V3 Pinout

aijayadams, 2019-12-12:
This wiki is linked from the product page as official documentation for the Wolf v3, but doesn't contain board diagram / pin out information about that product revision. Could we get permission to include (or link so) something like https://blog.seidel-philipp.de/impulserc-wolf-pdb-v3-fettec-kiss-fc-anschlussplan-wiringplan/

rauar, 2019-12-23:
I think a pin out diagram with descriptions of what the pins do would be better than Phillip's diagram. The latter highlights just a few specific use-cases and does so by drawing connections to other (specific) hardware. For example the vtx audio and mic pads and the cam control pad definitely lack documentation.

awolf78, 2020-02-01:
Done:
[Wolf V3 in Wiki](https://github.com/ImpulseRC/OSD/wiki/Hardware:-Wolf-PDB-V3)

closed 2020-02-01 by awolf78