ImpulseRC/OSD #46: OSX support

rauar, 2019-12-24:
I banned Windows years ago (last version I owned was XP). Just found out than on my Macbook I only can flash new firmwares. No configuration stuff at all via a nice UI😭.

Sijoma, 2019-12-24:
I agree. If you need dev support I would be willing to work on it myself. But unfortunately the OSD Configurator is not open source.

jadonbaker, 2020-02-03:
Also happy to assist with desperately needed macOS version.

awolf78, 2020-02-03:
We will most likely port our Windows GUI to Chromium this year - that depends on a few drivers that we need to make this happen. Those drivers/libraries are still in development and have not been stable enough yet for us to use.
Then we will support any platform that can run Chromium :)

MrMPod, 2020-02-21:
what is needed to install this on windows, there is security error and I am trying switch off my antivirus programs and nothing helps
I agree that GUI is needed for Mac too

awolf78, 2020-02-21:
Do you have local Administrator rights on the machine you are trying to install the Windows GUI?
Can you post a screenshot of the security error you are getting?

MrMPod, 2020-03-04:
I fixed the issue ... I had to setup record by regedit

HKEY_LOCAL_MACHINESOFTWAREMicrosoft.NETFrameworkSecurityTrustManagerPromptingLevel

and enabled some rows there and after that it is installing with no problem and works great