ImpulseRC/OSD #47: LED configurator

rauar, 2019-12-24:
Just added 32 leds to my quad. Cobfiguring them via OSD/Sticks will be a severe pain. Can we get something like in BF ?

Or at least some kind of "upload" option for a led config file ?

rauar, 2019-12-24:
PS: would be willing to help out coding something but in contrast to the previous OSD this seems to be closed-source now 🤬

awolf78, 2019-12-24:
Agree that the LEDs deserve GUI support. @sblakemore had some ideas he wanted to share with me how to make this nice and easy, but has not done so yet. #skeletonmeme

rauar, 2019-12-24:
Here my adhoc proposal based on how audio sequencers work:

- select an effect (e.g. no. 1-4)
- show a slider which selects a step in a sequence of e.g. 64 steps
- for each sequencer step render a grid of LEDs (grid size = sqrt(max allowed leds), each led is rendered using the color it's currently configured
- default color of each led is #000000, allow changing color code of each led by clicking on it (small popup with textfield and regex for input validation)

Not sure how the "internal" led data is stored but I would imaging that it's simply a huge array of
[effect][step][led] = <color>. That could be easily generated by the above when hitting save.

@awolf78 Any chance ImpulseRC is opening the "api" of the controller (at least) that we can build our own prototypes/scripts/tools around the Wolf PDB ? Personally I would be even happy with a self-written script instead of a GUI until a "standard" solution is available.