ImpulseRC/OSD #48: TBS Unify Race Pro V2 support

rauar, 2019-12-24:
TBS Unify Race Pro V2 has different power levels than the (already supported) Unify Pro.

Base Power is 25mW. Max Power is 200mW. I'm not sure if there are intermediate levels (don't think so).

200mW is not available in the OSD.

awolf78, 2019-12-24:
You mean this one?
https://www.team-blacksheep.com/products/prod:unify_pro_hv_race2_m

Or this one?
https://www.team-blacksheep.com/products/prod:unify_pro_hv_race

Can you post a screenshot or DVR? I do have a TBS UNIFY PRO 5G8 HV - RACE 2 connected to my development setup, it uses SA2.1 and transmits the various dbm levels it supports which are correctly displayed (25mW, 100mW and 400mW).

rauar, 2019-12-24:
It's the one on the first link you posted.

Oh - it seems the Race V2 hardware indeed does 25, 100 and 400mW (and not as I assumed 25 and 200).

I think the following link led me to think that the Race edition is designed to only power with 25 or 200mW:
https://team-blacksheep.com/products/prod:unify_pro_hv_race

... but it seems this is only valid for the Race V1. The race V2 seems to do 100 and 400mW as well.

That means the OSD is actually correct 😬

Sorry for the mix up.

closed 2019-12-24 by rauar