ImpulseRC/OSD #50: PitMode 0mW setting not available

rauar, 2019-12-29:
I tried putting my VTX pitmode on a switch. The docs state I should choose 0mW but the lowest power setting is 25mW for me. The TBS UNIFY PRO 5G8 HV - RACE 2 supports SmartAudio 2.1 so that should not be the issue.

rauar, 2019-12-29:
I think I figured it out. I think the docs might not be up to date and it looks to me like they describe a former way of activating some kind of workaround pitmode. As pitmode seems to be a dedicated feature now in the menus the 0mW instruction looks obsolete to me. Right ?

awolf78, 2019-12-29:
Since older SA based VTX (SA1.0 through SA2.0) could only enter pit mode during power up I initially setup the PIT BOOT concept - pit mode could only be activated during plug in (VTX hardware except for the Tramp would support otherwise anyhow) and you can leave pit mode, but not enter it again. SA2.1 devices support a 0mW mode and if you set 0mW as low power (or set it to an AUX channel) it is equivalent to having pit mode on a switch. The documentation is up to date, but unfortunately I have created a little bug when I adapted my SmartAudio code for the RushTank VTX so it won't show the 0mW anymore.

Many thanks for reporting this, will be fixed in next release!

closed 2019-12-29 by awolf78

rauar, 2019-12-29:
@awolf78 i'm currently thinking about pit boot and pit mode on a switch back and forth. i think there are 2 setups from a conceptual point of view:

a) pit boot on one switch + low power setting (25mW)/high power setting on another switch
b) no pit boot, only power settings switch (currently with 2 levels)

the advantage of a) is that you will not by accident kill the vtx transmission by going into pit boot power level (0mW) after launch. the disadvantage is that you occupy 2 aux channels with this setup.

b) only needs 1 aux channel but flipping the switch when in high power mode will end up in 0mW which might be fatal as well during flight 🙈

At the moment my setup is a) which I'm ok with (safest setup IMHO and I still have a few aux channels left at the moment).

You wrote you will bring back 0mW as low power setting. That means if I enter 0mW as low power setting I can only switch between 0mW and x mW. I would recommend to introduce a third level here and allow each level to be configured in terms of power output (as it's possible right now). additionally the lowest power level is dedicated for pit mode and can only be entered again if disarmed.

I would configure this e.g. as this (for my taste):

low: 0mW -> pit mode when not airborne
medium: 25mW -> default power (from a legal perspective)
high: xxx mW -> high power (personally I just would only and especially use this for emergency cases to avoid a crash due to vtx failure)

Those 3 levels could be nicely mapped on a 3-level switch and entering pit mode 0mW by accident is impossible.

What do you think ?

rauar, 2019-12-29:
ps: on a side note: i noticed last week that pit boot only became active *after around 8 seconds* when powering on. this behaviour is gone since today and I think the cause was that I (still) had a VTX configuration for the band/channel in the KISS FC (althout vtx wires are soldered to Wolf PDB). I suspect that the Wolf PDB noticed that something is configured in Kiss FC for VTX and that it allowed Kiss FC to win the race. I guess this that was causing vtx power to be 25mW for the first 8 seconds which renders pit mode/boot in this time frame useless (might cause fatal noise for other pilots).

awolf78, 2019-12-30:
The power and VTX settings are synchronized with the KISS FC as long as DummyVTX or higher is selected on the KISS FC (DummyVTX is recommended). The advantage of the synchronization is that you can use LUA to change you channel/band and also low and high power levels (or KKC). 
Due to the synchronization I like to keep it consistent and not differ from the KISS FC VTX settings.

Your argument with the AUX channels is not valid: Due to the configurable ranges you can use e.g. 3 position switches. PIT BOOT could be set to MEDIUM which can be the same as low power which could be on LOW+MEDIUM. High power would be set to high on the same AUX. Then you have all your VTX adjustments you want on one switch and AUX channel. 

I personally have PIT BOOT on MEDIUM of my arm switch which happens to be a 3 position switch - but that's a matter of taste.