ImpulseRC/OSD #53: Dead wolf pdb osd?

charalambos12, 2020-02-17:
Hello, im having a black screen. After a lot of checks, the vtx is working fine and the cam is dead. When i plug a new cam i continue to have black screen so i think its an internal video failure. Using a multimeter, the current output of vtx and cam pads is normal and the gnds are ok, so only the video pad remains. What do you think?
Setup:
Kiss v2
Wolf pdb v1
Unify pro hv

awolf78, 2020-02-17:
Even when the cam is dead or not connected, you should still see the OSD on a V1/V2. Is the red LED still blinking?

sblakemore, 2020-02-18:
@charalambos12 This issue tracker is really only meant for firmware related discussions. If you have a hardware issue can you please contact ImpulseRC support and they will help you troubleshoot.

closed 2020-02-18 by sblakemore

charalambos12, 2020-02-18:
Sure, sorry for that. Didnt know