ImpulseRC/OSD #55: Missing release notes / changelog

rauar, 2020-02-20:
Just wanted to check back here in the repo to find out if newer releases of the firmware are available and if so what was added/changed. But it seems there's nothing available here.

I guess the only option is to actually hook up the Wolf PDB to the laptop and start the ImpulseRC app (which is quite an act for simply checking if a newer firmware release is available).

awolf78, 2020-02-21:
I don't agree it is that much of an act to hook up the PDB and start the app (but that's opinions and matter of taste).
I do agree that there is no cumulative changelog anywhere and it is not very convenient in the app to browse through all the change logs. Will add them in the release section here.

awolf78, 2020-02-21:
Release notes are [here](https://github.com/ImpulseRC/OSD/releases)

closed 2020-02-21 by awolf78

rauar, 2020-02-21:
Thanks a lot for adding.

If you have no quad and/or a working laptop with the app installed and/or a usb cable at your hands (e.g. on the road, at work etc.) it is really difficult ;)

Again thanks !

PS: perhaps a release date makes sense as well 😇 at least for upcoming releases.

awolf78, 2020-02-21:
GitHub will take care of the release date automatically for future releases :)