ImpulseRC/OSD #56: OSD settings reset when removing battery

JohanJacobs, 2020-02-21:
Hi, 

Not sure if this is the correct place to post. 
when ever i unplug my battery all the settings on the Apex OSD resets back to default. (VTX channels and power, displayed items etc changes )
Even tried setting the defaults thru the PC software and through OSD.

setup :
Running ImpulseRC Apex frame
apex OSD and regulator 
Kiss FC v2, 
Johnny FPV camera  
tbs unify pro 5g8 hv race2

thanks.

awolf78, 2020-02-21:
Try flashing with Full Chip Erase using the Windows GUI. In the GUI, click Update. Once you see the update screen, make sure Full Chip Erase is checked and select flash.

JohanJacobs, 2020-02-21:
Thank you, 
update with Full Chip Erase selected fixed the issue :)

closed 2020-02-21 by JohanJacobs