ImpulseRC/OSD #57: ApexOSD + BF wiring unclear

udyux, 2020-02-25:
The [Betaflight WolfPDB](https://github.com/ImpulseRC/OSD/wiki/Betaflight-WolfPDB:-Please-read!!!) offers guidance on how to wire up the WolfPDB with a Betaflight FC (OSD RX to FC TX, BF settings, etc). 

The ApexOSD docs however offer very little in terms of wiring and offers no information concerning BF settings. 

My question is: should I follow the same instructions as the WolfPDB?

Might be helpful to others in the future to have this info either on the ApexOSD page or have a link to the Betaflight WolfPDB page. 

Thanks!

awolf78, 2020-02-25:
Yes, use the same instructions

udyux, 2020-02-25:
Thanks for the quick reply and all the great stuff. Closing the issue.

closed 2020-02-25 by udyux