ImpulseRC/OSD #60: MacOS - firmware flashing - app unavailable

udyux, 2020-03-17:
I'm having the scene cycling issue described in #34 on my RunCam Swift JohnnyFPV and intend to flash the latest firmware on my Apex OSD.

Trouble is when I follow [the app store link](https://itunes.apple.com/us/app/impulseosd/id1416313087) on the [ImpulseRC downloads page](https://impulserc.com/pages/downloads) I get a message saying "This app is currently not available in your country or region". For reference, I live in Canada.

For now I can set "CAMERA SYNC" level to `200` which should solve the issue, but are there any other methods for flashing?

awolf78, 2020-03-17:
The Swift JohnnyFPV needs a cam sync of 200.
We will look into the app store issue.

xayahrakan, 2020-04-13:
I have the same issue in germany. When clicking https://apps.apple.com/us/app/impulseosd/id1416313087  "This app is currently not available in your country or region".

udyux, 2020-05-23:
Closing as the problem seems to be resolved for me. Was able to install!

closed 2020-05-23 by udyux