ImpulseRC/OSD #61: Forgotten settings after a random number of packs

marcelkk, 2020-05-10:
I have the third Wolf PDB V3 now with the same issue: After a random number of power cycles (10-50 flights per example) the PDB starts and all settings are at the default values (OSD elements, VTX settings, battery...).
From this point the PDB only saves settings temporarily. I’m able to save settings at the begin of a pack, after power cycle it one time the settings are forgotten.

To Flash it with the GUI and „full chip erase“ will bring it back into normal working mode. But even then its a number of random packs again until the issue will start again.

6S Setup with 4x 470uF Low ESR Elkos (really well soldered and built well considered!)
Flyduino KISS V2 
Flyduino ESC 32 A
Runcam Phoenix 2/Caddx Ratel (same problem with both cams)
Unify Pro V3 5V

awolf78, 2020-05-11:
At this point I need:

- Detailed pictures how everything is wired up on your entire build (Not only the OSD related stuff)
- Backup of your settings from the Windows ImpulseOSD GUI.

marcelkk, 2020-05-11:
[image]
I hope the picture explains everything you need!?
[2020-05-11_14-34-51_ImpulseOSD.faa.zip](https://github.com/ImpulseRC/OSD/files/4609711/2020-05-11_14-34-51_ImpulseOSD.faa.zip)

Vegetabondfpv, 2020-05-11:
Hello, I have the same problem as marcelkk.

I have already discussed the problem with marcelkk and we both cannot understand what could be the problem.

My 4s lipo setup

Wolf Pdb v3 (softmount) 
Fettec F7 FC
4x 25A Kiss esc
TBS unify pro hv VTX
RunCam sparrow 2 pro
Crossfire nano Rx

[image]

awolf78, 2020-05-11:
Does this also happen when you flash V68.20?

marcelkk, 2020-05-11:
Yes, with both (V68.20 and V75.21)

awolf78, 2020-05-12:
Can you send me a backup of your KISS FC config also please?

marcelkk, 2020-05-12:
[kissfc-backup.txt](https://github.com/ImpulseRC/OSD/files/4616416/kissfc-backup.txt)
Sure!

awolf78, 2020-05-14:
You can try a beta if you want which will hopefully fix this issue (hard to find since I cannot reproduce it). Please post your MCU ID here so I can add you to the beta program. To get your MCU ID, press Ctrl+Shift+D in the Windows GUI and paste the contents of the Window that pops up here.
If you are on a MAC, go in the OSD menu (yaw left for more than 3 seconds) and go to SETUP->RESET. On the bottom of the Reset menu you will see a very long number - I need that including all the zeroes.

marcelkk, 2020-05-14:
I want to try everything that could fix it!

Device ID: 0047 7045 5547 7514 2032 2352

Is there any indication of what might be causing the problem? Please let me know if you're allowed to :-) For me it is also frustrating not to know whats wrong with my PDB/setup/whatever...
[image]

awolf78, 2020-05-14:
Please download and install the beta GUI from [here](http://osd-storage.impulserc.com/deploy-beta/setup.exe)

I added your MCU ID. Let me know how it goes.

marcelkk, 2020-05-15:
I flashed V79.23 and try it. I let you know if there are any problems. Thank you for the support!

awolf78, 2020-05-23:
I haven't heard anything so I am assuming this is fixed and I can close it?

marcelkk, 2020-05-23:
I would wait a few packs/flights more to be shure it’s fixed. In the past the failure happened after 4 weeks per example, so let’s wait a few weeks to be shure its really fixed. Until now everything is great 👌

marcelkk, 2020-06-04:
I think it’s time to say the issue is fixed!
No problem since flashing V79.23!
Thank you for the great support!

closed 2020-06-04 by marcelkk