ImpulseRC/OSD #63: SmartAudio v2.0 commands

slackista, 2020-05-11:
PDB v2, Unify 5V, 1uF cap between TX and mic.

Problem is why PDB is sending many packets to the vtx. According to
SmartAudio developers such implementation is wrong:
* at startup lots of packets (probing?) - wrong
* armed - one packet with increases power command - OK
* disarm infinity loop of commands - totally wrong

Startup: https://streamable.com/n3tr41
Arm and disarm: https://streamable.com/gp8owz

ImpulseRC support: "If your unhappy with any SA noise I would recommend using something with separate VTX control." - id est not our issue. Quite disappointing.

awolf78, 2020-05-12:
You could also solder the SmartAudio wire to your FC (KISS I guess) and from there a cable to the Audio on your Wolf or ApexOSD. Let me know if that improves your audio experience.

slackista, 2020-05-13:
Thank you for the suggestion. I will connect to KISS Tx5 instead of PDB.
Is SmartAudio VTX option in Kiss gui is fine with such configuration? Or interfering with ESC Kiss telemetry ?

awolf78, 2020-05-13:
KISS does ESC telemetry and VTX control on the same serial - no problem.

I have gone with a fine comb through my half duplex serial driver and smartaudio module - should be much quieter now. Will be a while until this hits a final release though, sending it to testers now.