ImpulseRC/OSD #65: Matek VIDEO TRANSMITTER 5.8G VTX-MINI

brian12568, 2020-05-26:
Hi,
Had this a couple of months got round to fitting it recently & managed to go fly yesterday.

The problem is the OSD Power value doesn't appear to change between arming/disarming it just flashes PIT however the LED's on the VTX does change to the power setting I have put in the OSD Gui when i arm the quad & then switch back on disarm so the protocol seems to be working at least for the actual VTX.

it seems its not been updated in the OSD

I am running the latest Firmware 

The Matek documentation says it uses the Tramp Protocol

awolf78, 2020-05-26:
Does it work if you don't use pit mode?

brian12568, 2020-05-28:
Hi, sorry I have only just seen this email.

I will check this evening

If you want to do some more active testing with me let me know.

Regards,
Brian