ImpulseRC/OSD #66: Can’t access Apex OSD menu in camera/ goggles

MJKitchin, 2020-05-26:
I have Impulse RC Apex Mr Steele quad. I have Foxeer Pred camera, Apex Osd, Kiss FCV2. I have been able to previously access osd menu using yaw left on Taranis X9d. Recently I was prompted in camera/ goggles to activate osd board and update osd via usb to impulse osd gui. I had activated and flashed to the latest firmware. I am now not able to access osd menu by yaw left on Taranis. On latest apex osd firmware update the red light remains solid, when I roll the firmware back to basics then the red light flashes. Either way, still no access to the osd menu in camera/ goggles like I was able to before. I have TBS Unify Pro Hv smart audio cable hard wired to Kiss FCV2 port TX5 and configured in Kiss GUI as always has been and worked this way before. 

Any advice to resolve accessing Apex OSD menu again? 

Thanks.

awolf78, 2020-05-26:
Try flashing with full chip erase.

MJKitchin, 2020-05-26:
I do not have an option on the Impulse RC GUI to 'Full chip erase'. I have followed steps of bridging SDA-GND, connecting lipo, removing bridge, connecting usb and GUI but no full chip erase option. Only firmware flashes.

awolf78, 2020-05-26:
Are you using Mac or Windows GUI?

MJKitchin, 2020-05-26:
Mac. I am guessing this is not possible on the Mac GUI. I do not have any windows.

awolf78, 2020-05-26:
When you say you "roll the firmware back to the basics": Do you mean you flashed a previous version? Or reset to defaults?
If you flashed a previous version: Which one is that?

MJKitchin, 2020-05-26:
I rolled firmware back one from v80... I still had a solid red light so tried rolling back to the bottom of the list. V1.1 and red light blinked as normal but still had no success with accessing osd menu, however, the ‘activate board’ message had disappeared from camera/ goggles. Flashing back to the latest firmware gives a solid red light and ‘activate board’ message appears again. I had never seen the activate board message before when I first built the drone. The message appeared for first time yesterday since I installed a new and the same Kiss FCV2. 

Yesterday I installed impulse osd app on Mac for very first time as message wouldn’t disappear. Since then flashing to the latest firmware, the message disappeared for one battery test flight, then today had reappeared and will not go away even after re-flashing. 

Unfortunately, I cannot see a ‘full chip erase’ on the Mac Impulse OSD gui. I am guessing this is a simplified gui version.

awolf78, 2020-05-26:
Unfortunately a Windows computer and Windows GUI will be necessary to fix this. Do you have access to a Windows computer?

MJKitchin, 2020-05-26:
Hi, I thought this might be the case. Unfortunately, I do not have access to a windows computer or windows GUI.

sblakemore, 2020-05-27:
Sorry, we have a new Mac GUI in development which we plan to be full featured, but that will some time before it'd ready. The only way to fix this at the moment is to find a Windows machine you can borrow for a short time, or install a trial of Windows in Bootcamp.

MJKitchin, 2020-05-27:
Hi all, thank you for your replies and advice on this matter. I managed to source a windows computer and fixed the issue quickly with the steps outlined for ‘full chip erase’. Thanks!

sblakemore, 2020-05-29:
Happy to hear you got it fixed! Hopefully we can have a nice Mac GUI for you fairly soon :)

closed 2020-05-29 by sblakemore