ImpulseRC/OSD #67: Critical Error

Paulson21, 2020-05-29:
When I plug my Windows computer into my Apex this is the error I get. I cant even use my Apex, sad I have a Mac and still can't use my Apex because you can only activate it via Windows(BS)...... My computer has the most updated .Net Framework. Tried two USB ports and two cables.... This is day three so do not ask me if I restarted my computer. Anyone capable of getting this far better know to restart a PC.
[image]

awolf78, 2020-05-29:
Try this: [Rescue flash](https://github.com/ImpulseRC/OSD/wiki/Rescue-flash)

MrMPod, 2020-05-29:
> When I plug my Windows computer into my Apex this is the error I get. I cant even use my Apex, sad I have a Mac and still can't use my Apex because you can only activate it via Windows(BS)...... My [...]

 I fixed the issue ... I had to add by regedit

HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\.NETFramework\Security\TrustManager\PromptingLevel

Paulson21, 2020-05-29:
Before you respond, I also have a brand new Wolf Board 3.0(XT60 Only), Does the same thing, fix your software. As a company do you realize people only own a mac? I have the luxury to have both but this is unacceptable that your products work on Mac, but you cant do the feature to activate the OSD.

Paulson21, 2020-05-29:
Wow so you have to be really computer savvy to edit your Regedit to make their software work..... What if I do not feel comfortable doing that? Impulse you need to update your software. This is inexcusable that a person with a windows PC only has to do a Regetit to make YOUR software work to USE your Products if they have this problem.

MrMPod, 2020-05-29:
> omputer savvy to edit your Regedit to make their software work.

I do not think that it is Impulse RC problem but Microsoft Software .... I am waitng for MacApp as much as You so till that time you can try ... it helped me and I am so happy with that OSD and frame Mr Steele style :)
"...
This error is caused by ClickOnce Trust Prompt Behavior. ClickOnce trust prompt to control whether end users are given the option of installing ClickOnce applications, such as Windows Forms applications, Windows Presentation Foundation applications, console applications, WPF browser applications, and Office solutions. You configure the trust prompt by setting registry keys on each end user's computer.

To better assist you with your concern, I'd like to know the following:

Confirm which build is currently installed on your device. To check, follow the steps below:
Press Windows key + R, then type in winver.
Click OK.
To adjust this settings, simply change the values in the Registry and you should be able to install the application:

1. Press Windows key + R, then type in regedit and press Enter.

2. In the Registry window, navigate to the directory below:

\HKEY_LOCAL_MACHINE\SOFTWARE\MICROSOFT\.NETFramework\Security\TrustManager\PromptingLevel\Internet

3. Change the value of the Internet to Enabled.

Note: The method to fix the issue will have Registry modification. Before you modify the Registry, make sure to back it up and understand how to restore the registry if a problem occurs. Please refer to this link: How to back up and restore the registry in Windows.

Some user had suggested that deleting the key TrustManager also fixes the issue. In addition, I'd recommend checking this article regarding how to configure the ClickOnce Trust Prompt Behavior.
..."