ImpulseRC/OSD #68: Apex regulator board - losing power to KISS FC

udyux, 2020-06-02:
The issue may be unrelated to the Apex regulator board but I figured I had nothing to lose by checking here if anyone has a clue. I've posted recently about a Holybro Kakute F7 setup but have since switched to a full KISS stack.

Maiden flight 2 days ago went well except that I hadn't set up the FC correctly to enable Airmode. I opted to go with the Idle Up approach based on a video Steele made about a year ago. Second flight, yesterday, ended after 50s. [Here is the DVR recording (volume warning)](https://streamable.com/a9wx1f), crash happens at 0:52.

Everything seems to indicate a loss of power to the FC just as I initiated a snap half forward pitch/half roll (lookback). ESCs were beeping out of sync after the crash but were beeping at a rate of 2/sec, indicating ESC failsafe. OSD was displaying alternating "No connection to FC 4" and "Telemetry lost" messages. Both the FC and OSD are powered via the regulator board and the OSD was still running after the crash. I also noticed that the OSD timer stopped counting the moment the quad seems to lose control but stays visible up until I hit the ground (which is when I would have disarmed).

I was able to reproduce these symptoms by arming the quad on the bench and pulling the Regulator/FC/OSD plug from the FC.

Setup:
ImpulseRC Apex frame running 4S
KISS FCF7 V2 [1.3RC40o]
KISS 4-in-1 ESC [latest stable]
Apex OSD [latest stable]
Apex regulator board
Foxeer Falkor V4
TBS Unify Pro 5G8 V3
Apex Hornet 2306.6 2555KV motors

It was built following Steele's build video, so the ESC is inverted and rotated and the FC is mounted on 7mm rubber standoffs. The wires are routed between the FC and ESC as he demonstrates but I have since noticed the tip in the docs concerning this.

Worth noting that I couldn't get my hands on the official ImpulseRC 3-way wiring harness for the Apex gear. I have a ton of pre-crimped JST-SH wires and connectors that I've used numerous times in the past and made my own harness. I figure I got the wiring right since the quad flew and I carefully followed the various wiring diagrams and used an image of the official harness as reference.

The SH plugs seem to be held in place by the pins and crimps, since placing an empty connector into a socket doesn't hold on its own. I'm going to try replacing the Gnd/VCC wires between the regulator and OSD but any other troubleshooting tips are more than welcome.

udyux, 2020-06-02:
Quick follow up question. There is very little mention of the pads on the regulator board in the docs.

If I want to eliminate the connector and regulator board as the source of the problem should I try simply soldering the VCC pad on top of the regulator board to the lipo positive on the FC and one of the LED grounds to the lipo negative on the FC?

awolf78, 2020-06-02:
The Apex OSD harness passes through Lipo voltage to the FC - so not regulated voltage.
For your case I would completely eliminate the ApexOSD and Apex Regulator. Not saying it has anything to do with your issue, but then it may be easier to pin point what the issue is. It may also be a connection issue. I would just keep the the ESC connection to the 4in1 ESC and pull separate wires to supply Lipo voltage to the FC. Good luck.

udyux, 2020-06-04:
As suggested I soldered wires straight from the 4S input to the FC. Flew several minutes over grass without issue, it waited for me to be over concrete this time before giving out. 

This build has drawn its max out of me. Months now and I still can't go 4 packs without it falling out of the sky, regardless of the stack I put in it.

Closing this issue since the one thing I do know now is it's not a connection issue with the Apex regulator.

@awolf78 Thanks for always being so responsive and patient with all of us. Class act!

closed 2020-06-04 by udyux