ImpulseRC/OSD #69: Issue with TBS GPS and apex OSD.

bosbefok, 2020-06-02:
I'm having an issue getting a connection with a TBS GPS. The GPS tx/rx pads are wired to the apex osd rx/tx pads. On power on the GPS receives power however the GPS related osd elements just continually flash. I've tried outside under clear skies for about 10minutes. I'm not seeing any updates at all, no satellites found, no coords etc.

Do you have any suggestions?

awolf78, 2020-06-02:
May have to wait longer than 10 minutes for first lock. Does the sat count go up in the OSD or does it stay at 0?

bosbefok, 2020-06-02:
It stays at 0. All osd GPS elements just flash continuously. Is there any way to get more info about whats going on?

awolf78, 2020-06-03:
Not really, but if it doesn't acquire any satellites for 10 minutes outside that is not a good sign. May want to try a different GPS module

bosbefok, 2020-06-03:
Tried with a brainfpv gps module and everything worked immediately. This indicates an issue with the TBS gps.

closed 2020-06-03 by bosbefok