ImpulseRC/OSD #70: Apex OSD no USB connection

krypticmindvhm, 2020-06-07:
Hi i want to update and get no usb connection, even not in windows .

it´s a broken connection or it´s a know fault?

Thx for help

awolf78, 2020-06-07:
Does the OSD work? Is the red light blinking?
If that is the case, do you have a different cable you can try?

krypticmindvhm, 2020-06-08:
Hi yes osd is working in fligth .
 
no theres no red ligth.
 
I tried several kables whose  workin normaly on fc or someting simelar :/
 
Greetings Goldmann

 
 
 
 

<https://github.com/ImpulseRC/OSD/issues/70#issuecomment-640214937> , or 
unsubscribe
"https://github.com/ImpulseRC/OSD/issues/70#issuecomment-640214937", "url": 
"https://github.com/ImpulseRC/OSD/issues/70#issuecomment-640214937", 
"name": "View Issue" }, "description": "View this Issue on GitHub", 
"publisher": { "@type": "Organization", "name": "GitHub", "url": 
"https://github.com" } } ]

awolf78, 2020-06-08:
Try this: https://github.com/ImpulseRC/OSD/wiki/Rescue-flash

krypticmindvhm, 2020-06-12:
Thx for help that worked !Von meinem Samsung Galaxy Smartphone gesendet.