ImpulseRC/OSD #71: WOLF V3 PDB OSD Black Screen

MVII3IV, 2020-06-19:
Hi guys
I have an issue with my WOLF V3 PDB, is totally new I have connected my Foxxer mini arrow and my TBS unify HV, when I connect the battery to my drone I receive no video, just a black screen... here comes the interesting thing, if I put my finger on the camera len and I block the light, the osd and video come back, but as soon as I remove my finger from the camera the osd goes away again...
If I leave the drone on the table and I turn off the light the OSD comesback... 
turns out the light is affecting the OSD I am new on Kiss and Wolf pdb, I dont know what to do....

Is this an issue or a configuration error???

awolf78, 2020-06-19:
Unfortunately the Foxeer mini Arrow delivers a very poor video signal that violates the NTSC/PAL standards severely. It is the only camera we could not get a decent sync on during our tests. Do you have a different camera you can use.
Sorry for the inconvenience.

MVII3IV, 2020-06-19:
Ohh I didnt know that, can you recommend me a good camera to work with?
do these cameras work?

- RUNCAM SWIFT MINI 2 MR. STEELE EDITION
- ETHIX Camera
- RunCam Swift Mini 2 - JohnnyFPV Edition

thanks for your quick response!!!

awolf78, 2020-06-19:
They should all work. The Runcam Swift Mini 2 (steele or JohnnyFPV version) may require a cam sync of 200.

MVII3IV, 2020-06-23:
thanks a lot!!!

closed 2020-06-23 by MVII3IV