ImpulseRC/OSD #74: Crossfire tx power missmatch value

slackista, 2020-07-22:
TX power label of Crossfire is showing '250MW' instead of '500MW'. Full Crossfire TX doesn't have option to transmit at 250mW.

awolf78, 2020-07-22:
I know, but unfortunately there is no way on my end to detect if it you are running full CRSF or micro. I don't receive the value it is transmitting at, just and index of the power level which then translates to a table. And that table is different on micro and full at that index (full has 500mW, micro 250mW).

slackista, 2020-07-22:
Check this out: https://github.com/BrainFPV/betaflight/blob/4064a2fb5668e9f2bb56aaee5310a90179c77217/src/main/rx/crsf.c#L203

awolf78, 2020-07-22:
Yeah, I can follow the same logic. This is not documented anywhere in the CRSF documentation and I cannot test this since I only have a micro TX (which switches to 7 on 250mW).

slackista, 2020-07-22:
I have full TX. I can test it :)

closed 2020-09-30 by slackista