ImpulseRC/OSD #75: Apex OSD mAh consumption value jumps by thousands

udyux, 2020-07-23:
I've already got a pretty good idea of what's causing the problem but figured, for posterity's sake, I'd open an issue in case I'm either wrong or someone else is experiencing the same behaviour.

The drawn mAh value will jump multiple thousands, eg. from 450 to 24000 in an instant. Seems to only occur on snap flips. Running the Apex OSD, latest FW, with the FETtec FC and the Flyduino 4in1 ESC. 

My theory is that the data is being corrupted by EMI from the ESC as I am running the telemetry wire under the FC. I'll be routing it over the top to see if it helps. 

Do with this issue as you see fit, I just thought it was worth mentioning.

awolf78, 2020-07-23:
If you have telemetry on your radio, you can actually narrow down if the issue is generated by bad telemetry data or the OSD. If the radio shows the same crazy value as the OSD it is due to the EMI you are suspecting (and me too BTW).

udyux, 2020-07-28:
I do have telemetry setup on my R-XSR, I'll have to look into how to get the readout as I've never really used it for anything other than RSSI.

My theory is drawn from the Flyduino FC shutdown issue I was having which turned out to be caused by running 4S with the Flyduino 4in1 mounted inverted (FETs facing up). A couple of us had the same issue with identical setups which was resolved by flipping the ESC back over. In my case the problem was happening on snap flips, as is the case with the OSD.

I'm waiting for my TX16S to arrive so I can finally make the switch to Crossfire which is when I'll try re-routing the wires, dont really feel like opening up the quad as it flies perfectly otherwise. I'll check the telemetry in the meantime.

udyux, 2020-08-10:
I've routed the Tx/Rx telemetry wires that link the FC and OSD on top, away from the ESC. After at least a dozen flights I can report that this seems to have resolved the issue.

Closing given the positive result.

closed 2020-08-10 by udyux