ImpulseRC/OSD #77: OSD Gone when armed

Paulson21, 2020-07-27:
Hello, hope everyone is staying safe. I love my apex, I have the Mr.Steele version.  I flew about 20 batteries with the stock settings not realizing my VTX was at 25mW. Once I switched my VTX to 500mW, my OSD is gone when the drone is armed.  OSD comes back when the drone is not armed. Did a full erase on the Apex OSD board and updated to the newest version available non beta. Problem still exist after all updates, It did work one time before I touched/changed any settings at all in the OSD. I just find it weird I flew over 20 flights with no issues and a VTX(Team Black Sheep Unify) made my OSD gone when armed now. Both ground wires (VTX and Camera) are on the same GND pad. 

Please let me know what I am doing wrong. Thanks for all the help in advanced.

awolf78, 2020-07-27:
Since you have a Mr. Steele Apex, I assume you are using the Apex Regulator board powering the Apex OSD. I also assume you have the standard wiring harness between ApexOSD, Apex Regulator board and KISS FC? 
If all my assumptions are correct: Try measuring the 5V on the ApexOSD (and 7.5V output on the Apex Regulator) when setting the disarmed power to e.g. 500mW. Let me know what the voltage meter says.

Paulson21, 2020-07-27:
Yes I am using the Apex Regulator. I am using the supplied wiring harness that came with the kit.

The voltmeter showed me 5.05V and 7.2V

awolf78, 2020-07-27:
Ok, what VTX model are you using (5V Unify?) and what camera? Is the + of the camera connected to 7.5V or 5V? What about the VTX?

Paulson21, 2020-07-27:
Team Black Sheep 5V Unify Pro 5V V2 soldered to the 5V pad.
Steele Swift Mini 2 soldered to the 7v pad.

awolf78, 2020-07-27:
Set the camera sync to 200.

Paulson21, 2020-07-27:
That makes the OSD distorted and jumpy

awolf78, 2020-07-28:
At this point I would need to see a DVR what exactly is happening when you arm the quad.

blackdrone911, 2020-09-12:
Hello,
I have a similar problem. Only that my OSD flickers during the flight and disappears sporadically, so that i can't see the telemetry datas. Also during landing and when disarmed, sometimes the OSD cannot be accessed.
What could I check?

Used component:
KISS FC v2
Wolf PDB OSD v3
KISS 32A ESC
TBS micro TX v2
TBS nano RX v2

blackdrone911, 2020-09-12:
Update,
seems that (for the moment) I solved the problem via setting Camera Sync.
But two question are open for me.
What is Camera Sync? It was setted to 0
Why I can do this setting only through the google and not in the Software (KISS, ImpulseOSD)?

[image]

awolf78, 2020-09-12:
You can set it in the ImpulseOSD GUI. Click on the 3 stripes on the top left and then on "Camera".

blackdrone911, 2020-09-12:
> You can set it in the ImpulseOSD GUI. Click on the 3 stripes on the top left and then on "Camera".

Yes thanks, rly stupid from my side. I got it!
But what does this value mean? And what’s the best “fitted” value?

awolf78, 2020-09-12:
It changes the synchronization level for the OSD and depends on your camera. Most cameras work fine at the default 0, some need 100, some 200. If you do not get any flickering and your OSD does not vanish then you are good and don't need to touch the value any more (keep it at 200).

blackdrone911, 2020-09-12:
> It changes the synchronization level for the OSD and depends on your camera. Most cameras work fine at the default 0, some need 100, some 200. If you do not get any flickering and your OSD does not vanish then you are good and don't need to touch the value any more (keep it at 200).

Ok! I got! Thanks a lot for the reply, that solved my problems!

nipponbike, 2020-09-22:
> Hallo,
> ich habe ein ähnliches Problem. Nur dass mein OSD während des Fluges flackert und sporadisch verschwindet, so dass ich die Telemetriedaten nicht sehen kann. Auch während der Landung und bei Deaktivierung kann manchmal nicht auf das OSD zugegriffen werden.
> Was könnte ich überprüfen?
> 
> Verwendete Komponente:
> KISS FC v2
> Wolf PDB OSD v3
> KISS 32A ESC
> TBS Mikro TX v2
> TBS Nano RX v2

same configuration -same problem : after 2 or 3 packs loose of OSD...have to disconnect an dconnect together

awolf78, 2020-09-22:
Try raising the camera sync to 100. If it still disappears, raise further to 200.

blackdrone911, 2020-09-22:
Yes exactly this helped me, thanks again!
Nipponbike take a look at the older posts. I am sure this will solve your problem.

nipponbike, 2020-09-30:
raise to 100 .....same problem, raise to 200 -totally loose of OSD ! - even the whole image disapears sometimes....  (need to connect the lipo a second or third time to get my image back) Is that PDB  broken ???

awolf78, 2020-09-30:
OK, try flashing with "Full chip erase". If that doesn't help I would need to see your wiring.

nipponbike, 2020-10-10:
Full Chip Erase did not help , same issue, it happened more with 6S than with 5s batteries...., but i´m done with that PDB...

nipponbike, 2020-11-01:
> OK, versuchen Sie mit "Full Chip Erase" zu blinken. Wenn das nicht hilft, müsste ich Ihre Verkabelung sehen.
Sorry, the Wolf_PDB is OK !!! , but this PDB is definitive not compatible with a CADDX Ratel-Camera !  I switched back to my RuncamSwift-Mini and everything works well !!! Thank you !