ImpulseRC/OSD #79: OSD Jumps around and will not save RSSI settings since switching to CRSF

udyux, 2020-08-17:
I've been flying for the last few months without any issue using the following equipment:

- ImpulseRC Apex with FETtec F7 KISS FC & Flyduino 4in1 25A ESC (running 4s)
- Apex regulator & OSD boards
- TBS Unify Pro 5G8 V3 & Foxeer Falkor 2 mini
- FrSky R-XSR receiver

As of yesterday, I finally have a Crossfire Micro TX V2 and installed a CRSF Nano RX in my Apex. Two problems have appeared since:

1) The OSD jumps around on the screen ([short clip of the issue](https://youtu.be/PGmwSuXWeDY)).

2) When I change the RSSI setting on the OSD and set it to CRSF+LQ preset and save+exit, it doesn't seem to save it. This is without a power-cycle or anything. Just save+exit, go back and it says "Custom"

Just to be clear, the only change since this started is swapping from FrSky to CRSF.

I've flown 7 packs so far, updated my OSD to the latest version, changed the VTx channel, all with no change.
My KISS FW is `1.3RC45y` and my Crossfire TX/RX is running `v3.75`.

awolf78, 2020-08-17:
1. Set the camera sync to 100. That should fix the jumping OSD issue.
2. Assuming you have 40o or higher on the FC: Leave the RSSI setting on "AUTO" (Defaults). It will pick up the CRSF data automatically and set it up appropriately.

udyux, 2020-08-17:
Thanks I'll give that a try and report back. 

Another thing I noticed was that I could hear the SmartAudio clicking for much longer than before. Don't know if it's related. 

Any idea why this might have started at the same time as making the switch to CRSF?

udyux, 2020-08-19:
@awolf78 I've set the camera sync to 100 and tried again to set the RSSI to auto. Just to confirm, this was happening before and after I updated the OSD FW to the latest version. I also noticed all my settings had been preserved after the update, not at all a complaint but wasn't sure if this was supposed to be the case.

**Result:** the flight OSD no longer jumps around however the OSD menus and temporary labels like "armed", "disarmed" or quick battery selection have started glitching.

As for the RSSI, at the end of the following clip you'll see me try to change settings in the RSSI menu. Every time I set the preset to CRSF and come back the preset has reverted to "custom" and even with the custom mode selected the only value I can change is the AUX channel, the rest of the settings change for split second and go back the the value it was at. Is this the expected behaviour?

[Video link](https://youtu.be/NCLziZDWYlA)

There are two other issues since I swapped from FrSky R-XSR to the TBS Crossfire Nano RX:

1) As mentioned in my previous comment, the SmartAudio clicking sound when I arm and disarm (probably to change the power level) goes on for much much longer than before. It used to give two clicks consistently. You can see what I mean when I disarm in the video.

2) I've started seeing faint black lines in my feed when I hit the throttle which wasn't something I'd been seeing before. You can see them at the very start of the video and also when I touch down and theres a small blip of the throttle before I disarm. From my understanding this is usually power related but I seem to recall seeing somewhere that excess D-gain can produce similar video banding. Nonetheless, last time I changed my tune was long before switching to Crossfire.

For reference I've been running the VTx at 800mW. Maybe also worth noting is that the receiver is mounted on top of the Apex regulator board with double sided tape (as was the R-XSR) and is installed exactly as Steele mounts his in his build video (with the excess antenna lead held down under the receiver).

I'm a bit baffled by the fact that switching to Crossfire could introduce these kinds of issues but I've literally not changed anything else on the quad.

awolf78, 2020-08-19:
RSSI/LQ settings: Leave them as they are. When the RSSI channel is on "AUTO" you cannot change the other values. On "AUTO" it configures everything automatically and grabs the CRSF LQ information automatically from the KISS FC. You don't have to do anything, your LQ is displayed properly in the OSD and it barely drops below 100% unless you start flying a few km out or fly behind objects (careful). 

Video glitches: The cam sync of 100 seemed to have helped, but judging by the video you may need to go higher. When you change the value in the OSD it is applied immediately. If you go up and it starts jumping more - go down. Try to find the minimum and maximum value where it starts to get worse and then set it in between. Then you should be fine overall.

SmartAudio:
I literally had the same experience on my own quad. But I did not change anything on the quad, all I did was open it and put it back together. The SmartAudio communication is not the most stable and moving the wires a little bit can cause interference - then it needs a few more tries to make sure the command got through to the VTX. 
Keeping the wires very short and making sure they are routed away from any possible should improve this.

Black video lines/noisy video:
This is a known issue with the Fettec FCs and may require some re-routing of the grounds. Can you give me a short description how you have it wired up along with some pictures?

Change from FrSky to CRSF:
To be honest I have no explanation why you are having all these issues where you had none before. I can only point to the known grounding issues related to the Fettec FC. For some reason those did not show up when you were on FrSky. Possibly the change in amperage draw from the FrSky Receiver to the CRSF receiver had an effect on this.

udyux, 2020-08-19:
Thanks for the detailed response @awolf78, staying true to ImpulseRC's "above and beyond" mantra as always! I suspected that some of my issues were unrelated to the Apex PDB/OSD so thanks for taking the time to share your insights.

**RSSI**
Ok, reassuring words.

**Video glitches**
I'll try calibrating this and let you know which number did the trick for future reference.

**SmartAudio**
Makes sense, I've got the VTx mounted to the top plate above the camera/OSD just in front of the FC. The wires are already quite short, I often tug on them with the top plate in hand before removing the connector. I noticed yesterday that the OSD 5V  and GND to the VTx seemed weak at the joint. To be safe I replaced both wires and did a better job on the joint.

I have all VTx and camera wires neatly tucked in between the camera and the OSD board. We'll see if there's improvement since changing those two wires.

**Black lines**
I'll see if I can get some pictures after work but I haven't done anything fancy. For clarity, I'm running FETtec FC with Flyduino 25A 4in1 ESC. I'm running 4S with no cap.

I couldn't find the ImpulseRC OSD/PDB wire harness in stock anywhere but have a bunch of pre-crimped wires and made one. Wiring:
- 8 pin connector under FC to 6 pin connector on ESC (last two slots on FC plug are empty - GND / VCC)
- ESC LiPo pads feed 2x VCC and 2x GND to the PDB
- 7.5V, GND and LED wire running from PDB to OSD connector (routed _beneath_ FC)
- VCC and GND going from PDB to connector on top of the FC
- RX/TX from same FC connector as above to OSD connector (routed _above_ FC)
- GND pad on OSD is shared by camera and VTx
- 7.5V, Video in and Cam control from OSD to camera
- 5V, Video out and audio/SmartAudio (bridged) to VTx

Not that I think it will make a difference but the Nano RX is wired to the connector beneath the FC. When I first wired it up I had inverted the TX/RX wires. Only trouble it seemed to cause was no control received by the FC until I swapped them.

awolf78, 2020-08-19:
Try running the "7.5V, GND and LED wire" above the FC. 
If I understand this correctly, the FC is powered and grounded by the Apex Regulator Board (you call it PDB) ONLY. There is no (direct) GND between FC and ESC?

udyux, 2020-08-19:
Yeah, I get tired of writing regulator haha. Ok for the regulator -> OSD harness.

I though I was thorough yet the only detail I left out is the only one that mattered! There is a GND between the two on the ESC -> FC harness. The last two on the FC are empty as mentioned, but the 6 others are motors 1-4, Telemetry and GND.

awolf78, 2020-08-19:
Ok, lets see if moving the wires above the FC helps you. If it doesn't, the next step would be to remove the GND between Apex Regulator and FC and add a GND between FC and Apex OSD (any GND pad on both sides works).

udyux, 2020-08-20:
After routing the regulator/osd wires over top I tugged on the VTx wires by accident and the audio/SA wire came loose. Naturally I replaced it. 

I don't know if it made a difference but I'd read in the past that twisting video and GND can help shield a bit. So I twisted the SA, Vid and GND to the VTx together. 

When I plugged in, without making any changes to camera sync, everything looked good in the menus! 

So for the jittering menus either routing the wires over the fc, twisting the signals and GND, fixing a weak joint or some combination of these resolved that issue. 

I'll have time to fly after work so I'll see for the black lines and SA clicking. I'll also try putting the sync back to 0 to see what happens.