ImpulseRC/OSD #80: GPS not working

dizzy32141, 2020-08-20:
Im really having a hard time to get my GPS to work on my Wolf PDB + KISS V2 build. 
I bought a M8 GPS from TBS about 2 Months ago. 
It worked on an older Version of KISS and WOLF PDB but then it stoped working someday. At first I didnt realize it.
All I did was updating KISS and WOLF PDB till now but I never got it to work again.
So I thought the GPS may be defect and bought a Matek M8Q-5883. 
Soldered it on the PDB and got the RED and Blue LED on the GPS but nothing on my OSD! 
I checked the wires with a multimeter, nothing wrong. 
I double checked the wiring and also soldered on different 5V, TX, RX and GRD pads. 
Then I also noticed that my quad does not double peep like usually anymore after I plugged my battery in. 
So something must be wrong but I dont know what it could be.  
On the Wolf PDB there is a green light with a flashing red light (should be normal i think). 
I can arm it normally but no GPS in OSD and no peep before start anymore (tested with and without GPS). 
The OSD shows only the compass but its only flashing.
Please let me know what I possibly did wrong or if there is a defect or issue with the firmware or something.
I really dont know what to do anymore. I attached some photos and videos. Please excuse my bad english. have a good day

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

https://youtu.be/ZB6Eq6Xg9vs 
https://youtu.be/r5Zh2EjeMVg

awolf78, 2020-08-20:
Please swap the RX/TX wires on the PDB. RX(GPS) goes to TX(PDB) and vice versa.

dizzy32141, 2020-08-20:
Swapping the wires solved it! Also got a double peep now after reconnecting the LED Stripe! Thanks a ton man!