ImpulseRC/OSD #82: Add support for NEO-M9N

slackista, 2020-10-12:
Add support for new ublox chip NEO-M9N to take advantage of its full potential.

awolf78, 2020-10-12:
Have you seen any commercially available boards yet for the M9N? That fit on a 5"-7" mini quad?

slackista, 2020-10-13:
After ~two weeks we should able to buy Matek M9N-5883 for $45. Size and weight is affordable for 7".

slackista, 2020-10-17:
https://store-drotek.com/917-DP0801.html

slackista, 2020-11-02:
There are now plenty of stores with NEO-M9N. Does current firmware support UBX-NAV-PVT messages?

awolf78, 2020-11-02:
I'm still waiting on mine to show up in the mail. I may add a 25Hz mode if it works reliably.

slackista, 2020-11-20:
To be clear, NEO-M9N not working with latest firmware v86.25

slackista, 2020-11-24:
I received my neo-m9n and is working fine with v86.25 latest firmware. I've got 16 satellites in big city after one minute cold start. Closing that issue.

slackista, 2020-12-13:
Reopening issue. Mine M9N sometimes dropping satellites to LOW count. Same behavior when I chose setting `used`(GGA max 12 satellites) or `available` in gps settings. Such issue can cause rth drop. DVR presenting issue: https://streamable.com/d88hcv

awolf78, 2020-12-13:
If you post your MCU ID here (press Ctrl+Shift+D in Windows GUI) I can give you a beta version you can try. That will also allow to run the GPS at 25Hz (instead of 10Hz).

slackista, 2020-12-13:
Sure. Device ID: 0023 302F 5534 4507 2032 2339

awolf78, 2020-12-14:
I added your MCU ID, please use this link for the beta GUI: [Beta GUI](http://osd-storage.impulserc.com/deploy-beta/setup.exe)

awolf78, 2020-12-21:
Have you tested the beta? Did you try with 25Hz? Does RTH work on 25Hz?

slackista, 2020-12-21:
Haven't got time yet to test. I have beta firmware on 7" now, so probably I would not test RTH with 25Hz on it. To heavy in case of drop. For sure I will test 25Hz flying and 4 GNSS systems simultaneously. Probably on this Sunday.