ImpulseRC/OSD #84: OSD cell voltage & weird behaviour

dizzy32141, 2020-12-27:
I was testing my 6s Li-ion battery and had a little crash were one of the crossfire antenna went off.
I didt'n notice it at first but after another flight with a normal 6s pack  I went indoor and replaced the antenna.

After replacing I went out and flew another 6s Li-ion pack. I noticed that the cell voltage in the osd was completely off. 
It showed me a value between 4.20 and 4.40. 
I was very confused because only if I plugged in the Li-ion batterys, the value was wrong. If I plugged in a normal and uncharged Lipo the value was correct. 
I reflashed the FC, the OSD and the ESC but the problem persists. I reseted the telemetry data in the taranis but in the taranis the values are correct.  I charged the Li-ion batterys and the cell voltage was correct again. But everytime I connect a uncharged Li-ion battery it gives me a wrong value in the OSD. 
Is it because of the low voltage of the Li-ion, a bug or is anything broken? 

Then a flew a pack to test if its safe flying. The cell voltage is ok until I plug an empty Li-ion battery in. But now I had massive OSD flickering till the OSD was completely gone. I set the camera sync to 100 and it fixed the issue.

Now everything is ok except the stats of my CRSF-link and the timer shows no symbols anymore. Its just a black box were the symbol normaly is. Here's a screenshot:
https://ibb.co/89HJ20F

Could you please tell me what is causing the black boxes and the cell voltage thing? Thanks man!

awolf78, 2020-12-27:
Please flash your OSD again with "Full chip erase".

dizzy32141, 2020-12-27:
I flashed the OSD with full chip erase but the black boxes are still there. Do I have to do a downgrade first? I noticed that after flashing the OSD, I have to update the Impulse OSD Logo that appears at startup. If I dont update the logo, its like one big black box instead of the logo at startup. 
Here's a screenshot: https://ibb.co/KrYLmnN

awolf78, 2020-12-27:
Hm, if you flash with full chip erase it does write the default font completely new and all those black boxes should disappear. There is a possibility for a defect on you OSD board. Please contact your dealer - it is up to them how to handle this situation.

krunked-fpv, 2020-12-27:
Have him do the rescue flash where u solder the pads with a wire and snip
it.  Worked for me.

awolf78, 2020-12-27:
> Have him do the rescue flash where u solder the pads with a wire and snip it. Worked for me.

That only forces the bootloader so you can flash in case the firmware got corrupted somehow or cannot start.
The OP has a different issue, something is going wrong during flashing or he forgot to click the "Full chip erase" button. If Full chip erase does not help the STM chip on his OSD board may be defective...

dizzy32141, 2020-12-29:
Thanks guys! Today I will rescue flash just in case and if its not normal after flashing I probably will use this PDB in my freestyle quad. What do you think? It's safe to fly like that?
I also have a broken PDB V3 here, were a solder pad went off. What about replacing the STM Chip from this old board?

awolf78, 2020-12-29:
After you flash with "Full chip erase" make sure you leave the Lipo connected for a minute or two. It rewrites the font and if you unplug the Lipo during that time these issues can happen.
If the board that is giving you issues is also a PDB V3 you can swap the STM chips if you feel comfortable doing that.

dizzy32141, 2021-01-02:
Hi Im back. Tried to full chip erase but my device is blacklisted now. I tried 5 times or so before. Still need to try it with lipo plugged in for a few minutes. 
Device ID:  https://ibb.co/4dWxrkh

awolf78, 2021-01-02:
Removed you from the blacklist.

dizzy32141, 2021-01-04:
So I did the rescue flash and full chip erase properly. No improvements but I will use it in my freestyle quad then. Thx for the great support and have a good one guys