ImpulseRC/OSD #86: Font not changeable + missing signs

blackdrone911, 2021-02-21:
Hey!
1. Why it’s not possible to change the OSD font? It’s shown in the GUI but in the googles it’s still the basic font. 
2. Also there are some signs missing. Instead of °C you can see there 2 big black bars. The same you can find in the overview when disarming the quad. 
How I can fix that?
Rescue I tried 2 times without any solution for both problems.
[image]
[image]

awolf78, 2021-02-21:
Please flash again with Full Chip Erase. IMPORTANT: After you have flashed (and clicked OK to go back to OSD setup screen in the GUI), keep the Lipo plugged in for another 20-30 seconds. During that time it flashes the font again. Then the characters you are missing will be shown correctly.
Font: Can you show me how you are doing that? In a video?

blackdrone911, 2021-02-22:
ok now it looks much better! Now there is only a black bar in the picture but we are making progress!
Attached is the video, but I think you can’t do something wrong there. Finally this is working too!
[image]
[image]

awolf78, 2021-02-22:
OK, looks like the font changes accordingly. 
The black bar in front of the (MCU) temperature should also have been fixed by the full chip erase method - you can try again and wait after it restarted. There should be a small square box with white legs that looks like a MCU when the font is correctly loaded.

blackdrone911, 2021-02-22:
I tried 3-4 times after I got tired. Tomorrow I’ll try to flash the old software than the newest one, maybe this helps?

blackdrone911, 2021-02-23:
So, now the PDB is blocked after several full chip erases. What now?
[image]

awolf78, 2021-02-23:
Post your Device ID here - or if you do not feel comfortable doing that open a support ticket with ImpulseRC. They will unblock it for you.

blackdrone911, 2021-02-24:
Ok, here is the image with the ID.
Is there another method the get the bugs out instead of full chip erase?

[image]

awolf78, 2021-02-24:
I removed your device from the blacklist.
I am guessing it is only that one character next to the temperature that has that issue? The only way to fix that is with a full chip erase, but since you have done this multiple times already and it did not help I can only assume that this part of the flash on you chip is defective. You can present this case to your dealer, but it is up to them how to handle this.

blackdrone911, 2021-02-24:
Nah, it’s nearly new :)
So, now it’s done! Seems that the downgrade helped me out! Everything looks as it should! Great, I am fine again! Many thanks for the support!

[image]

closed 2021-02-24 by blackdrone911